arendst/Tasmota #1239: problems with DS18B20

ppisljar, 2017-11-23:
Hello,

i am using sonoff basic, latest version of sonoff tasmoa (from today) and have the following setup:
- 3 DS18B20 sensors connected to GPIO14

connections:
- 3.3V as power source (first pin on sonoff basic)
- GND (4th pin on sonoff basic)
- DATA (5th pin on sonoff basic)
- 4.7k resistor between DATA and 3.3V (just one for all 3 sensors)

nothing in the log (like sensor busy or anything), seems the sensors are just not detected

i also tried moving it to GPIO1 (serial in) and GPIO3 (serial out) with same results.

however i am sure the sensors function, as i tried using ESP32 with this code:
https://github.com/DavidAntliff/esp32-ds18b20-example/tree/master/main

and i get all 3 sensors detected, and read, with 0 errors (same connection, 3.3v power source from esp32)

should i connect DS18B20 in a different way for sonoff tasmoa ? or what would be my next debugging steps ? 

thanks.

ppisljar, 2017-11-23:
oh, and i compiled the code by uncommenting this line in platformio.ini:
```
env_default = sonoff-ds18x20
```

no other changes to the code.

davidelang, 2017-11-23:
what version of tasmota are you running, and did you configure the GPIO to the 
sensor type in the module configuration page?

ppisljar, 2017-11-23:
as mentioned in my first post, i downloaded tasmoa from github today (master branch), and yes, GPIO is set to 4 (ds18b20)

arendst, 2017-11-24:
No problem at all:

[image]

[image]

[image]

You might want to try to connect only one sensor and disable USE_DS18x20 in user_config.h to see if one sensor works fine using default enabled single senosr support.

arendst, 2017-11-24:
Try the latest development repository version 5.9.1i.

ppisljar, 2017-11-24:
i tried with a single sensor today, compiled without USE_DS18x20 ... also doesn't work.
i'll try with development branch tomorrow.

pascalsaul, 2017-11-25:
Looks good to me also:
tele/dual_sw/SENSOR {"Time":"2017-11-25T13:54:49","DS18x20":{"DS1":{"Type":"DS18B20","Address":"28FFE48601170396","Temperature":24.2},"DS2":{"Type":"DS18B20","Address":"28FFDE8D0117033B","Temperature":22.7}},"TempUnit":"C"}
stat/dual_sw/RESULT {"OtaUrl":"http://192.168.1.12/sonoff.bin"}
stat/dual_sw/RESULT {"Upgrade":"Version 5.9.1f from http://192.168.1.12/sonoff.bin"}
tele/dual_sw/LWT Offline
tele/dual_sw/LWT Online
cmnd/dual_sw/POWER (null)
stat/dual_sw/UPGRADE Successful. Restarting

tele/dual_sw/LWT Offline
tele/dual_sw/LWT Online
cmnd/dual_sw/POWER (null)
tele/dual_sw/INFO1 {"Module":"Sonoff Dual","Version":"5.9.1i","FallbackTopic":"dual_sw","GroupTopic":"sonoffs"}
tele/dual_sw/INFO2 {"WebServerMode":"Admin","Hostname":"dual_sw","IPAddress":"192.168.1.122"}
tele/dual_sw/INFO3 {"RestartReason":"Software/System restart"}
stat/dual_sw/RESULT {"POWER1":"0"}
stat/dual_sw/POWER1 0
stat/dual_sw/RESULT {"POWER2":"0"}
stat/dual_sw/POWER2 0
tele/dual_sw/STATE {"Time":"2017-11-25T13:56:43","Uptime":0,"Vcc":3.322,"POWER1":"0","POWER2":"0","Wifi":{"AP":1,"SSId":"<ssid>","RSSI":82,"APMac":"78:C1:A7:3D:B0:B8"}}
tele/dual_sw/SENSOR {"Time":"2017-11-25T13:56:43","DS18x20":{"DS1":{"Type":"DS18B20","Address":"03170186E4FF","Temperature":24.0},"DS2":{"Type":"DS18B20","Address":"0317018DDEFF","Temperature":23.2}},"TempUnit":"C"}

ppisljar, 2017-11-27:
i tried with latest development version, and a different sonoff basic (just to be sure something is not wrong with the first one). still no success ....

in the meantime i tried to connect DHT22 sensor, and this one seems to work ok.

ppisljar, 2017-11-27:
also tried with version 5.8 (with and without #define USE_DS18x20 ) ... no luck ...
at this point i am almost sure something is wrong with the connections or my sensors are not compatible ...

ppisljar, 2017-11-27:
i tried with smaller resistor, and this seems to work ... yay (3k) ... atleast for a single sensor ... lets try with multiple now :)

pascalsaul, 2017-11-27:
I use 4.7k resistors with 1 or even with 5 DS18B20's...

ppisljar, 2017-11-27:
yup, it works now. in case anyone else runs in the same problem ... i tried with smaller resistor, which solved my problem. the code referenced above (for esp32) works with the original resistor as well. my cable is 1m long.

ppisljar, 2017-11-28:
ok i am getting into more issues, i hope somebody here can advise me ...
i tried to connect the 3 sensors over 10m CAT5 cable now
- blue/lightblue: data
- orange/lightorange: vcc (3.3v)
- brown/lightbrown: gnd
- green/lightgreen: unconnected

i am using two 4.7k resistors, one on each end of the cable ( i also tried with single 4.7k resistor as well as single 2k resistor)

the readings on all 3 sensors are 85* most of the time, from time to time jumping down a bit to something like:

DS18B20-1 Temperature | 78.9°C
DS18B20-2 Temperature | 85.0°C
DS18B20-3 Temperature | 75.4°C

reading around internet, for example here: https://forum.arduino.cc/index.php?topic=20574.0  or here: http://www.protonbasic.co.uk/showthread.php/60921-DS18B20-Long-Wires

makes me think that 10m should not be a problem (its actually a bit less, more like 7m)

the first link mentions something about timing, but i haven't tried changing the timings yet, i am wondering if anyone here is using longer cables and has good results before doing that.

using latest version of tasmota firmware 5.9.1h

ppisljar, 2017-11-28:
i shortened the cable to the minimum i need (4 meters), still the same issue

mkh595, 2017-11-30:
If sensor returns 85 (default value in scratchpad after power up before first conversion is triggered) it means that conversion and writes to internal scratchpad doesn't finished succesfully. It can be problem with timing, means that your sensor is bad quality and needs more time for conversion. Try another sensor or you have to use logic analyzer to check comunication.
I have few sensors which I throw away because permanently returns 85.

Frogmore42, 2017-12-04:
There are lots of things that can go wrong with OneWire communication and DS18x20 devices.  I have been using them for years and they are generally pretty reliable, but they can be tricky.  I have never tried a 10m cable, but I have 5m or more.  I typically use the solid orange wire for 3V3 or 5V (on other controllers that support it), the solid brown wire for ground, and the white with brown stripe for the data line.  I chose this because the data is in a twisted pair with ground (which seemed like a good idea) and it uses the pairs that are not typically used in normal CAT5 T568B wiring for network or phone (which seemed like an even better idea in case they accidentally got plugged into a phone/network port.

In my code, I throw out readings of exactly 85 or 0, since either could be caused by bad power or a short (both of which I have seen, due to poor soldering and/or corrosion over time).

Are you using raw parts or ones that already have 1m or longer cables attached?  I stopped making my own sensor cables and now buy the premade ones in stainless steel casing with 1m or longer cables.  I do typically solder the provided cables to CAT5 cable (only the flexible kind used for patch cables, not the in-wall stuff which uses solid wire and is too easy to bend and break) to make longer cables.

I have not tried long cables on the esp8266 so I don't know how well it can handle them, but check your soldering first.

ppisljar, 2017-12-05:
thanks @Frogmore42, seems my problem was bad contacts. I have sensors with 1m wire, and then i used CAT5 cable to extend them, but i didn't solder them together, used the screw kind cable joints (no idea how to properly call that in english). 

I soldered them now and it seems to work OK.

closed 2017-12-05 by ppisljar

Frogmore42, 2017-12-06:
I use solder and then heat shrink tube if they will not be subject to moisture.  For ones that will be outside, I encase them in epoxy or hot melt glue.

comtechnik, 2018-02-02:
I have a problem with the DS8B20 on Sonoff RF.
The DS18B20 gets hot and sonnoff does not start anymore.
I have connected to Pin1, GND and GPIO14, between Pin1 and GPIo14 is a 4.7k resistor. Who can help me ??? Thank you

lobocobra, 2018-09-27:
Does anyone know if using a DS1820 does work?
=> I can connect a DS18S20 to my Sonoff Basic and it works. But none of the DS1820 I tried does show a signal. I have a 10kOhm resistor and a 1m long wire.

As said the "S" version works. Before I return the 10 DS1820, I wanted to check if somebody can confirm, that they do not rund with the latest standard firmware?

I ordered the DS18S20 and they sent me the DS1820. Does somebody knows the answer?

Thanks

lobocobra, 2018-09-27:
Solved the issue myself :-) 2  out of the 10x I bought were defect, guess which of the 2 ones I tested ^^
=> all others work.

Frogmore42, 2018-09-28:
Here is an article on the differences
https://www.maximintegrated.com/en/app-notes/index.mvp/id/4377

Depending on the resolution you have selected, you might see differences in temperature.  The higher the resolution the more likely there will be a difference shown.  It is rare for the temperature to be exactly uniform.  There is usually some difference/gradient.

dhlii, 2019-02-04:
I have a DS1820 with a Sonoff-Basic and Tasmota 6.4.1(basic)
Data is connected to GPIO14. 
Power and GND connected. 
A 4.7K resistor between power and data. 
In configure Module I have "GPIO14 Sensor" set to DS18x20(04)
Should I see any evidence Tasmota found the DS18x20 via the web interface ?
In the console Log ?

dhlii, 2019-02-05:
The firmware is sonoff-basic.bin from the git repository. 
I did not build firmware. 
I already have weblog 4 on and there is no mention of a DS1820 or anything 1wire in the log.
I checked the resistor with a VOM before soldering it between +3.3 and GPIO14
I have only a single sensor/DS1820

Should I create a new topic ?

ehorohorin, 2019-02-05:
@dhlii try another ds18b20 if you have one more

edgiru, 2019-03-22:
I look not only at one does not work for me. Yavro problem in the firmware. Also worth 4.7k resistor. but does not see the sensor. other firmware (espeasy) immediately see the sensor.

ascillato, 2019-03-22:
You have upload the version sonoff-basic.bin.
Please read the release notes. The sonoff-basic.bin DOES NOT HAVE SENSORS. 

Please, for sensor use sonoff-sensors.bin

Thanks 

If you need further assistant, please find us in the Tasmota support chat. Thanks