arendst/Tasmota #1241: Support for Electrodragon ESP LED Strip Board

canis-ferrin, 2017-11-23:
I am looking for a 5050 LED Stripe driver board and found Electrodragon ESP LED Strip Board as described here: [http://www.electrodragon.com/product/esp-led-strip-board/](url) 
Is this compatible with Tasmota?

arendst, 2017-11-23:
Select the wemos d1 mini and the correct pwm channels as documented on their site and you should be  good to go.

Alternatives H801, Arilux and MagicHome are already supported.

closed 2018-05-07 by stale[bot]

empenoso, 2018-11-11:
> Select the wemos d1 mini and the correct pwm channels as documented on their site and you should be good to go.
> 
> Alternatives H801, Arilux and MagicHome are already supported.

I bought ESP LED Strip Board, Mosfet Drive https://www.electrodragon.com/product/esp-led-strip-board/. Board works with Sonoff-Tasmota. RGB LED Strip also works.
But do not work together.

I can't send "+12V" to exit "V". https://www.electrodragon.com/w/ESP_Light

Please tell me. Thank!