arendst/Tasmota #1242: Unable to flash sonoff-ds18x20 when it is below upload_resetmethod in platformio.ini

Frogmore42, 2017-11-24:
I have a generic Wemos D1 Mini (with the CH340).  It was working fine when flashed from one computer, but when I moved it to another computer it would give me this error:

|   |-- <ESP8266WiFi> v1.0
Compiling .pioenvs\sonoff-ds18x20\src\sonoff.ino.o
Linking .pioenvs\sonoff-ds18x20\firmware.elf
Checking program size
text       data     bss     dec     hex filename
462447     8816   36576  507839   7bfbf .pioenvs\sonoff-ds18x20\firmware.elf
Building .pioenvs\sonoff-ds18x20\firmware.bin
Looking for upload port...
Auto-detected: COM3
Uploading .pioenvs\sonoff-ds18x20\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
==================================================================== [ERROR] Took 22.37 seconds ====================================================================

Since it worked fine on the other computer I tried to figure out what was different.  Turns out, none of the configurations that are listed below the line that says "upload_resetmethod" would work.  But, if I moved them above that point it works fine as it should.

I am running this on Windows 10 Creators Edition (and also the Fall update 1709).  In all cases the compilation works correctly (after failing the first time I change things, but that is a different bug).  Anyone else seeing this?

arendst, 2017-11-24:
As designed. You will have to update platformio.ini yourself with the correct upload information as seen under [env:sonoff].

General Platformio knowledge.

Frogmore42, 2017-11-25:
If you mean, "you have to uncomment 'upload_resetmethod = nodemcu' when using a wemos or nodemcu device", I got that part and did that.  If you mean, it is general Platformio knowledge that you have to have some specific order of things, I did not get that and I am not sure exactly how I would get that knowledge.

I did figure out what the problem was and just reported it as an issue, because I am pretty sure other people have run into the issue too (based on my search of the issues).  But, I am guessing that most people are use an FTDI connection and the reset and GPIO 0 button to get into boot loader mode so don't run into the issue.

A simple reordering of the file looks like it would solve the problem.

arendst, 2017-11-25:
I don't get it. What do you re-arrange to make it work for you?

Frogmore42, 2017-11-25:
I moved this block
`[env:sonoff-ds18x20]
platform = espressif8266
framework = arduino
board = esp01_1m
board_flash_mode = dout
build_flags = -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=707 -DUSE_DS18x20 -DMESSZ=600
lib_deps = PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON, OneWire`

above this section
`; *** Serial Monitor options
monitor_baud = 115200

; *** Upload Serial reset method for Wemos and NodeMCU
upload_resetmethod = nodemcu
`
It made little sense to me that this would work, but I figured I would give it a try (since I had already tried changing the names (switch sonoff and sonoff-ds18x20).

I took a closer look at all the issues and it appears no one else has seen this particular issue.  Perhaps no one else is using Wemos (have not tried this on NodeMCU boards yet) or perhaps this is new, or perhaps it is just on Windows.

Looking more closely at platformio.ini, your first assessment might have been correct and the change I made might not be the best.  What would probably have helped me figure out what was needed would have been to have these lines in every [env] block:
`; *** Upload Serial reset method for Wemos and NodeMCU

;upload_resetmethod = nodemcu

; *** Upload file to OTA server using SCP

;upload_port = user@host:/path

;extra_scripts = pio/sftp-uploader.py

; *** Upload file to OTA server using HTTP

;upload_port = domus1:80/api/upload-arduino.php

;extra_scripts = pio/http-uploader.py

`
This would have made it obvious that each configuration needs to have an upload reset method (and I assume the rest of the stuff).  I'll read more about PlatformIO to better understand how it works.

Frogmore42, 2017-12-06:
Looks like the issue has been resolved with a recent change.

closed 2017-12-06 by Frogmore42