arendst/Tasmota #1243: Desktop Alexa App / Smartphone App will switch all Devices ON

zibous, 2017-11-24:
Whenever I use the Desktop Alexa App "Smarthome" section then all Sonoff switches are turned on. 

https://alexa.amazon.de/spa/index.html#appliances

Also when using on the smartphone, the same happens when I select the device in the application. It is not possible to turn on / off the application and the status "On / Off" is not displayed correctly.

Is this a mistake in the amazon alexa application or can this be prevented by the devices (settings) ?
This does not apply to all other smarthome devices (Philips Hue Lamp, Tplink Switch HS100).

zarss, 2017-11-25:
I had similar problem, changed to Belkin WeMo single device or Hue Bridge multi device an is fine now

zibous, 2017-11-26:
@zarss

Thanks, is working.
Only that the Sonoff switch will be shown as lamp. No nice but no problem.

reloxx13, 2017-12-15:
Heya, since my pull request did not get merged till now, u can get my WeMo fix from my fork: https://github.com/reloxx13/Sonoff-Tasmota/tree/feature/Fix_WeMo_Emulation

Alexa is requesting the status of the modul, which turned the device back on.