arendst/Tasmota #1247: Wemos D1 mini ws Wemos D1 pro

zarss, 2017-11-25:
I found this cool project not so long ago. Big thanks to people who made it. 
I have couple Wemos D1 mini running, but one of them is quite far away from router, so decided to swap it with Wemos D1 pro. Tried on two of them, but they are not working. My router don`t see them. Is the programming routine the same as for mini? Or i need to make any changes? Thank you.

ionciubotaru, 2017-11-26:
Try to move Wemos D1 pro near the router and check if works.
- if works is a hardware problem with antena, try to reposition the router antenas or the wemos pro device
- if id doesn't work at all it is a esp problem or a programming problem. 

I am sure that Tasmota is ok, because all wemos, sonoffs, use the same esp8266 chip.

zarss, 2017-11-26:
So. Normal Wemos D1 mini are ok. I have couple of them working. But Wemos pro are visible on the router, they have IP addresses assigned, but cannot connect to the web interface. If i try arduino web server sample from IDE, they work fine. What i am missing here?

smulle48, 2017-11-27:
D1 mini pro support external antenna. maybe it's an option for you :-)

zarss, 2017-11-27:
Oh silly me. Did not know, that you can use them without external antenna.

smulle48, 2017-11-29:
If you Got a laptop, try to connect to your WeMos with serial monitor when located where it’s not working. Then you get some info about the problem :-)

mkh595, 2017-11-30:
External antena is not supported by default. You have to re-solder resistor to select way to external antena
[How to use external antenna on Wemos D1 mini pro](http://raspi.tv/2017/how-to-use-external-antenna-on-wemos-d1-mini-pro-surface-mount-rework-video)

zarss, 2017-12-01:
Ok, thanks for clue about antennas. That part is done now and tested, boards are still alive. Tried couple examples on all my pro boards, all of them working. Uploaded Tasmota on them - not working. Nothing comes up in serial monitor. Tried D1 mini with the same build, everything works fine. Is there anything else i can try?

smulle48, 2017-12-01:
@zarss I have the exact same problem with a new Wemos D1 Mini I just received.

I found a workaround :-)

Flash this to your mini/mini pro:

https://automatedhome.party/2017/01/06/wifi-garage-door-controller-using-a-wemos-d1-mini/

Start the serial monitor to see it's starting (ip: 192.168.1.4)

connect to the AP GarageESP and setup wifi.

When done, look in the serial monitor for the URL Webupdate. open that link and upload the pre-compiled tasmota firmware. DONE :-)

I don't know why, but it worked here :-D

arendst, 2017-12-01:
The problem with Tasmota might be related to flash mode DOUT.

Perhaps the wemos d1 pro does not work with DOUT as it uses a 16MB flash that might not support DOUT. If you try another flashmode you will run into problems next time you try to update as Tasmota forces to flashmode DOUT as soon as the code is OTA or upload upgraded

Until further notice (and your experience) you are on your own regarding Wemos D1 Mini Pro and Tasmota.

zarss, 2017-12-01:
@smulle48 tried your way. After 5-8 attempts managed to run it on my device, could get into it using web interface, but not working after restart. So i guess, time to give up. Thank you everyone who tried to help.

mkh595, 2017-12-01:
For Wemos D1 mini/Pro I am using mode DIO and don't have problem, also OTA works

smulle48, 2017-12-01:
Mini Pro have to be DIO

arendst, 2017-12-01:
Did anyone try DOUT?

zarss, 2017-12-02:
Could you please explain, about what are you talking? How can i use mode DIO? Lammer here.

closed 2017-12-02 by zarss

mr-const, 2018-09-21:
I tried to flash with DOUT and succeeded, however had to bump SDK version from 2.3.0 to 2.4.1 because after first reboot ESP went crazy after each flash attempt.