arendst/Tasmota #1249: Support of Sonoff Dual R2 variant

vanyid, 2017-11-25:
Hello Tasmona community,

I have purchased 3 x Sonoff Dual modules. But those are different from any other mentioned module. (or just I not found anything on the web, maybe my fault) Does anyone have the same board?

The label on the board: **Sonoff Dual R2**. Ver 1.0 , 2017-7-20 (maybe the manufacturing date)
Inside the case there is only one chip a **ESP8285**.
The relays and the LEDs are directly connected to the ESP.

I will open a pull request according to the new module as soon as I am able to check in the necessary modifications to my fork repo.

Here is the pinning of the ESP:
This variant is more simpler then the previous one. (and perhaps cheaper for iTead)

PIN | Function
-- | --
GIO0 | BUTTON 0 (inside header)
GIO5 | L2 RELAY
GIO9 | BUTTON 1 (inside header)
GIO10 | SWITCH (on the case)
GIO12 | L1 RELAY
GIO13 | BLUE LED (inverted)

Some pictures about the board:

[image]

[image]

[image]

[image]

davidelang, 2017-11-25:
since you have found what's connected to what, you should be able to use the 
wemos module type and define the various GPIOs

vanyid, 2017-11-25:
As far as I see no it will not work 100%. Because the Button1 and the case switch are connected to the GPIO9 and 10, but in the "WeMos D1 mini" the GPIO 6 - 11 are not defined as GPIO_USER.

And otherwise in my opinion if this is an official Sonoff device then is should be supported by default.
I do not know when will iTead update the official wiki page and provide the schematic for this variant?

arendst, 2017-11-27:
The R2 will be included in Tasmota as a seperate device.

For now I use your findings regardiong GPIO and add it to the next (pre)release.

ThomDietrich, 2017-11-27:
I hate to be that guy to always push the documentation...
I'd suggest to keep the documentation up to date and one article that definitely would benefit from the images and a few details posted above is: https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual  (see for an [example for two versions](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro)) @vanyid maybe you could invest a few minutes there?

smadds, 2017-12-01:
The best thing I can see in this new version is that they have finally stopped using the fuse clips that were quite incapable of handling 16A. Have a look here for tales of woe when trying to use the fused Sonoffs with high currents: http://support.iteadstudio.com/support/discussions/topics/11000012976?page=1

closed 2017-12-01 by arendst

ThomDietrich, 2017-12-01:
@arendst still no documentation. Way to go 💯

localhost61, 2018-04-16:
For documentation I've annotated previous ESP8285 picture with pinout issued from the Espressif datasheet. I was looking for available GPIO pins. It appears that GPIO 2, 4, 6, 7, 8, 14 & 16 are free but only available on the ESP8285 chip, which requires some squills to solder. Instead of ADC (pin6) it seems that it's GPIO15 that is used to monitor the power supply.
In fact the easiest way to connect a sensor will be to use RX and TX on the header. The sensor will have to be removable in case of serial reprogramming, but I guess not in case of OTA update..

[image]

cpfarher, 2018-08-20:
@vanyid I have the same board, How do you flash tasmota firmware? I have the arduino uno, it is the same process as esp8266 but instead of select 8266 generic board I need to select generic esp8285 module?... It is compatible for add external switches?
Thank you

localhost61, 2018-08-24:
@cpfarher I hope you already found your answer in the wiki, [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE)

pmknowles, 2018-09-22:
Has anyone managed to get an AM2301 to work on the Dual R2. I have tried Button 0, Button 1 ,Tx and Rx and just get NaN as a response.(v6.2.1). Have tried other versions back as far as 5.12 and the sensor doesn't appear in the UI after configuration.

Bersaker, 2020-06-04:
Hi. I've a new Version with Layer date from 2019-09-04

[image]
[image]
[image]
[image]

localhost61, 2020-06-04:
Yes I got 4 of them too and they behave the same as v1.0

Bersaker, 2020-06-05:
Is it possible to use a bme280 over sda and scl?  When I look at pcb 1.4 pin 14 is in use

pmknowles, 2020-06-05:
Could you use the Rx and Tx pins at the top of the board?
Need to be careful at boot but should be ok.

Regards

Phil K

localhost61, 2020-06-05:
Yes, no problem to use the header on the top of board to connect a BME280 but you'll have to disable SerialLog (it's maybe automatic in Tasmota code).
The delicate point is to use Button 0 (GPIO9) which will prevent reboot if pull down.

[image]

[image]

pmknowles, 2020-06-05:
I really wish people would read this page.
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
It’s not just GPIO0 that causes reboot problems. Tx (GPIO1) will also cause a boot fail if it’s taken low as will GPIO2

Regards

Phil K

[2020-06-05 08_34_06-D11 VR_SdB2 - Configuration du Module]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F33861984%2F83845079-7dc71f00-a708-11ea-91c7-31c8ffc315ca.png&data=02%7C01%7C%7C82d121a72cc84877e67a08d8091d1339%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637269367759298939&sdata=ra7LR5ZG1uJ%2FALC7A98bMbBAb85%2BfSwyUv9k9QdcvtQ%3D&reserved=0>

[2020-06-05 08_35_47-D11 VR_SdB2 - Menu principal]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F33861984%2F83845154-96373980-a708-11ea-8c2b-9470d5b10ccd.png&data=02%7C01%7C%7C82d121a72cc84877e67a08d8091d1339%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637269367759308935&sdata=8P1xleVpqAautHPRNpJkbeJT%2FF4uns1n6LtEgHIIwIY%3D&reserved=0>

meingraham, 2020-06-05:
@pmknowles That's a great summary. You should add it to https://tasmota.github.io/docs/Peripherals/#usable-pins

pmknowles, 2020-06-05:
Done

Phil

kugelkopf123, 2020-06-05:
The same exists for ESP32. Perhaps this would be great to add it, too.

https://randomnerdtutorials.com/esp32-pinout-reference-gpios/