arendst/Tasmota #1252: Restarts when the internet can not connect..

salihed, 2017-11-26:
Hi,
I want to use tasmota on a device I developed. It works nicely when the internet is connected. However, when there is no internet, it restarts continuously. I tried to find a solution by looking at the wiki. I took a hard reset using Esptool. I changed the CFG_HOLDER value. But it did not happen. The relay turns on and off when you restart. I do not want this to happen. Could you help?

The loop in the serial monitor:
�
00:00:00 Project SmartNode_Test Sonoff (Topic SmartNode_Test, Fallback DVES_0ABA41, GroupTopic SmartNode_Test) Version 5.9.1i
00:00:00 WIF: Connecting to AP1 G4_9588 in mode 11N as SmartNode_Test-6721...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 atdnet_RPT in mode 11N as SmartNode_Test-6721...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
(....)

salihed, 2017-11-26:
The source of the problem is wificonfig. When I make wificonfig 4 from console, it does not restart. When I do WPSConfigactive, it starts restarting again. How can I solve this problem?

smulle48, 2017-11-27:
Remember to close the issue :-)

closed 2017-11-28 by salihed