arendst/Tasmota #1253: Switch is not reported via mqtt

boaschti, 2017-11-26:
Ist it a normal behavior that tasmota don't report a Switch immideatelly if it is pressed?

Client mosqsub/32281-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (15 bytes))
{"POWER2":"ON"}
Client mosqsub/32281-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/sonoff/POWER2', ... (2 bytes))
ON
Client mosqsub/32281-raspberry sending PINGREQ
Client mosqsub/32281-raspberry received PINGRESP
Client mosqsub/32281-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/sonoff/RESULT', ... (16 bytes))
{"POWER2":"OFF"}
Client mosqsub/32281-raspberry received PUBLISH (d0, q0, r0, m0, 'stat/sonoff/POWER2', ... (3 bytes))
OFF

The switch is reported periodically but not immediatelly.

Client mosqsub/32281-raspberry received PUBLISH (d0, q0, r0, m0, 'tele/sonoff/SENSOR', ... (63 bytes))
{"Time":"2017-11-26T17:44:32", "Switch1":"OFF", "Switch2":"ON"}

ionciubotaru, 2017-11-27:
set `switchtopic` and try different `switchmodes` - Tasmota reports switch changes via mqtt instantly.

boaschti, 2017-11-27:
I tried it. 
The problem is: 
If pysical switch ==Off and homeassistant sends to cmnd/sonoff/Power On then the switch (which is read from hass) is set to on. 
Perhaps  is another way to configure that.

smulle48, 2017-11-27:
mine is publishing the same time I press the button in the web interface / firming mqtt command. Try to subscribe to # (all) to see all messages.

When I used HASS, I used this code to control my sonoff:

switch:
  - platform: mqtt
    name: "Lampe Stue"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true

Now I have abandoned HASS in favor of node-red :-)

boaschti, 2017-11-27:
Yes that`s the config to controll relay but if you want to know the real switch state (in hass binary_sensor) sonoff don't tells it immidiatelly. You can only see the "output" on topic stat/sonoff/POWER. And if you set the switchtopic = sonoff you get same msg on cmnd/sonoff/POWER.

boaschti, 2017-11-27:
In my case i have a pir sensor which is connected to port configured as switch1 the relay1 switches the lamp switchmode1 = follow. Additionally i have a switch in hass. If i switch the lamp on with hass i also want to know if pir is on or off because it controlls other lamps too.

stefanbode, 2017-11-28:
I double checked on my devices. You need to define a RELAY1 to get immediate feedback through MQTT. The switch itself is not reported through the RESULT event. You just get it indirect through the POWER result or by waiting for the STATUS

boaschti, 2017-11-29:
Hm. Ok then aktual there is no way to get the real switch state immeditelly. Just indirekt (Power) but it can be inverted... Is it possible to implement it in the next fw version?

davidelang, 2017-11-29:
if you set switchtopic, then changing a switch doesn't change the relay, it only 
sends a mqtt message. Set switchmode to define how to respond to the switch

boaschti, 2017-11-29:
Ok. But i think the relay don't switch when any switch topic is set. So in this case i have to link switch and relay in hass. Right? I'll try again. Switchmode have to be 1 (follow).

davidelang, 2017-11-29:
On Wed, 29 Nov 2017, boaschti wrote:

> Ok. But i think the relay don't switch when any switch topic is set. So in this case i have to link switch and relay in hass. Right? I'll try again. Switchmode have to be 1 (follow).

well, when it switches the relay, you should get an immediate stat/ message that 
the relay has changed.

But I thought the problem was you didn't have a relay defined for that switch?

In any case, your home automation software can get the message and instruct 
anything to happen when it gets the message

boaschti, 2017-11-30:
Yes thats right i get the stat of relay if it changes. But if not like in the upper explained case the Relay doesn't change and i cannot see the switch state. switchmode1 = 1
pir Sensor = off -> switch lamp on with hass (cmnd/sonoff/power1 ON) -> i get the state via mqtt (stat/sonoff/power1 ON)

pir changes to on -> i get no msg no relay change because it is already on but i want to know that pir is changed to on

pir changes to off -> relay turns off -> (stat/sonoff/power1 OFF)

> In my case i have a pir sensor which is connected to port configured as switch1 the relay1 switches the lamp switchmode1 = follow. Additionally i have a switch in hass. If i switch the lamp on with hass i also want to know if pir is on or off because it controlls other lamps too.

smulle48, 2017-11-30:
Why not configure the pir as switch 2 with switchtopic pir and let Hass do the switching with the relay?

mkh595, 2017-11-30:
In Hass I have switch and PIR and both are reported via mqtt immediately.

[image]

PIR on GPIO1, external button on GPIO3 (standard GPIO0)

[image]

SwitchTopic: "home208pir"
ButtonTopic: ""
SwitchMode1: 1

```
binary_sensor:
# 208 - PIR
- platform: mqtt
  name: home208_move
  state_topic: "cmnd/home208pir/POWER1"
  payload_on: "ON"
  payload_off: "OFF"
  device_class: motion

switch:
# 208 - Light
- platform: mqtt
  name: "home208"
  state_topic: "stat/home208/POWER"
  command_topic: "cmnd/home208/POWER"
  payload_on: "ON"
  payload_off: "OFF"
  qos: 1
  retain: false

davidelang, 2017-11-30:
I believe that switchmode 7 (just added) will signal any state change of the 
switch.

But in your case, you need to have the pir set to something other than switch1, 
or you need to set switchtopic to decouple it from the relay

What you are seeing is what is intended, if you have the relay connected to the 
switch and you manually turn on the relay, the swtich being turned on isn't 
supposed to do anything.

it's doing what you've asked it to.

If you want it to do something different, then you need to change how it's 
configured (switchtopic) so that the switch isn't directly tied to the relay.

smulle48, 2017-12-02:
I have made a garage door opener with a reed switch. Finally I got it working as it's supposed to.

The reed sensor is set as a switch2, with switchmode2 7 and switchtopic garagealarm.

every time the reed switch is changing, it's publishing the message: cmnd/garagealarm/POWER2 ON or cmnd/garagealarm/POWER2 OFF depending on the current reed switch state.

This should also be possible to do with your PIR sensor :-)

boaschti, 2017-12-03:
@smulle48 yes it works but there is no funktionality without server

@mkh595 right but the Problem is explained upper you read your relay state and it can be differ to switch

@davidelang switchmode has to be 1 because relay have to follow the switch

So i think i'll decouple relay from switch in tasmota an couple it in hass. Not the best solution but it works.
I'll add a immediatelly switch update to the firmware. Who can commit it for me?

boaschti, 2017-12-03:
Adding 
`        if (MqttShowSensor()) {
          MqttPublishPrefixTopic_P(2, PSTR(D_RSLT_SENSOR), Settings.flag.mqtt_sensor_retain);
        } `

in sonoff.ino Line 2110 will fix that. 
can anybody commit it?

closed 2018-05-07 by stale[bot]