arendst/Tasmota #1254: Sonoff TH16 external pushbutton on GPIO 04 not working.

spacefolder, 2017-11-26:
Hi Guys! I have access to GPIO 04 on a TH16 (soldered to the available pad). I can use an external switch without problems, but I can not use it a a push button. (Button 2, 3 and 4 don't seem to work) How can I achieve this? Thanks, and cheers!

spacefolder, 2017-11-26:
If through the web console I issue command switchmode3 or switchmode4, when I check the switchmode (empty) I get the default settings, as if it is not taking into account my command..

22:55:48 CMD: SwitchMode4
22:55:48 MQT: stat/sonoff_alacena/RESULT = {"SwitchMode4":0}
22:55:53 CMD: SwitchMode
22:55:53 MQT: stat/sonoff_alacena/RESULT = {"SwitchMode1":0}

Any clues?
Cheers!

davidelang, 2017-11-27:
switchmode4 is setting the mode for swtich for

I think you are intending to do switchmode 4 to set switch one into mode 4

spacefolder, 2017-11-27:
Hi David, thanks for the reply..

That´s what I attempted to do (see previous post), but the setting is not "sticking". Not I´m not sure I´m entering the command correctly. This is what I´ve done.. 1) Web Console 2) Enter command SwitchMode4, 3) Check switchmode by entering command SwitchMode, and it defaults back to mode 1. What is the correct way of doing this?

(From the previous post)
22:55:48 CMD: SwitchMode4
22:55:48 MQT: stat/sonoff_alacena/RESULT = {"SwitchMode4":0}
22:55:53 CMD: SwitchMode
22:55:53 MQT: stat/sonoff_alacena/RESULT = {"SwitchMode1":0}

Cheers!

arendst, 2017-11-27:
If you have configured GPIO4 as a switch it is probably configured as ``Switch1`. In that case you need to use command ``switchmode1 <option>``. There is no switchmode4 as there is no relation between GPIOs and commands but between the number of the switch and the commands.

spacefolder, 2017-11-28:
Thanks @davidelang & @arendst 

After experimenting for a while, "switchmode 3" did the trick!
 
It seems I was entering the command incorrectly. The wiki says "SwitchMode<x>", so I wasn't leaving a space between the command and the value.

I hope this is useful for anyone trying to use an external wall switch (pushbutton, like a doorbell) with the same functionality as the internal sonoff button.

BTW, Remeber to set GPIO 04 to 09 Switch 1
[image]

Cheers!

davidelang, 2017-11-28:
On Mon, 27 Nov 2017, spacefolder wrote:

> That´s what I attempted to do (see previous post), but the setting is not
> "sticking". Not I´m not sure I´m entering the command correctly. This is what
> I´ve done.. 1) Web Console 2) Enter command [...]

note that you are entering the command 'switchmode4', which is asking for the 
mode of switch 4

if you want to set the mode of switch 1 to 4, then you need to send the command 
'switchmode 4' note the space there.

davidelang, 2017-11-28:
switchmode2 3 sets switch 2 to mode 3

switchmode without a number defaults to switch 1

so switchmode 3 sets switch 1 to mode 4

but just switchmode3 is a query to ask what mode switch 3 is in

closed 2018-05-07 by stale[bot]