arendst/Tasmota #1255: Sonoff Basic GPIO4 restart problem

jannnfe, 2017-11-26:
Hello,
i want to use two Reed Sensors on one Sonoff Basic. So I found out, that I can connect second Reed Sensor to GPIO4. The first ist connected at GPIO14.
The Problem is, that every time i trigger the second Reed Sensor on GPIO4 the Sonoff reboots and reboots and reboots on his own.
Maybe I have a soldering issue? Please have a look at this picture:
[image]

Thanks for your help :)

gajotnt, 2017-11-27:
Well it looks like really bad soldering.
You joined several pins on the chip...

ionciubotaru, 2017-11-27:
try gpio 1 or 3 (tx/rx) instead of gpio14 - is easier to solder

jannnfe, 2017-11-27:
I thought it was my soldering. but that was really hard to get such a small pin ran. I will try to completely remove the solder joint.
Thanks for the information with RX and TX. I did not know that this is possible. Of course that is much easier.
Thanks everyone!

closed 2017-12-31 by jannnfe

gajotnt, 2017-12-31:
https://github.com/arendst/Sonoff-Tasmota/wiki/GPIO-Locations