arendst/Tasmota #1257: Web Admin Password twice

don-willingham, 2017-11-27:
Typically, systems require a user to enter a password twice to ensure they match. If the user mis-types it, they may be locked out if they can't guess the typo, and I would assume they would have to re-flash their device. Two of mine are mounted in hard to reach places.

arendst, 2017-11-27:
You can always change the password using mqtt as it is only used for http.

don-willingham, 2017-12-08:
That's a good point that I didn't consider. However, not all users integrate with MQTT.  I also feel the user interface should also require the current password. Currently the password is exposed in the "view page source". I'll work on making the change at https://github.com/don-willingham/Sonoff-Tasmota/tree/issue_1257, as I assume this will be desired by others. However, do you feel the majority would prefer it the way it works now? Should it be conditionally compiled in?

arendst, 2017-12-08:
If it takes more code it should be optional as http is not core business.

closed 2018-06-23 by stale[bot]