arendst/Tasmota #1258: HTU21D detected with I2Cscan, but no readings...

RobKuipers, 2017-11-27:
Hi all,
I've connected a HTU21D to a Sonoff Basic with the latest Tasmota (5.9.1), but I get no readings of temperature and/or humidity in the webconsole or MQTT. 
It is detected with I2Cscan in the console though. 
I have restarted the module several times, both using the console (restart command) and by unplugging the mains. I've also let it rest for a while, but to no avail.

Any thoughts as to what I might be doing wrong?
Regards, Rob

Logfile:
00:00:00 Project sonoff DVES_98468D (Topic deuropener, Fallback DVES_98468D, GroupTopic sonoffs) Version 5.9.1
00:00:00 WIF: Connecting to AP1 H in mode 11N as deuropener-1677...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on deuropener-1677.local with IP address 192.168.11.31
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/deuropener/LWT = Online (retained)
00:00:08 MQT: cmnd/deuropener/POWER = 
00:00:08 MQT: tele/deuropener/INFO1 = {"Module":"Sonoff Basic", "Version":"5.9.1", "FallbackTopic":"DVES_98468D", "GroupTopic":"sonoffs"}
00:00:08 MQT: tele/deuropener/INFO2 = {"WebServerMode":"Admin", "Hostname":"deuropener-1677", "IPAddress":"192.168.11.31"}
00:00:08 MQT: tele/deuropener/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/deuropener/RESULT = {"POWER":"ON"}
00:00:09 MQT: stat/deuropener/POWER = ON
10:42:37 CMD: i2cscan
10:42:37 MQT: stat/deuropener/RESULT = **{"I2CScan":"Device(s) found at 0x40"}**

stefanbode, 2017-11-27:
Quite interesting. I have the same issue with the ADS1115. I assume an Error in I2cRead16. Function deliver not TRUE with the right address. Will do some investigations

stefanbode, 2017-11-27:
You can overcome this issue in support.ino if you change the response from 0 to 1. The problem, is, that the device sends a "0" as an answer.

With the small addition: if (data == 0) {data = 1;} you can quick fix it.
@arendst : This is a bug that might be fixed differntly 

```
int32_t I2cRead(uint8_t addr, uint8_t reg, uint8_t size)
{
  byte x = 0;
  int32_t data = 0;

  do {
    Wire.beginTransmission(addr);             // start transmission to device
    Wire.write(reg);                          // sends register address to read from
    if (0 == Wire.endTransmission(false)) {   // Try to become I2C Master, send data and collect bytes, keep master status for next request...
      Wire.requestFrom((int)addr, (int)size); // send data n-bytes read
      if (Wire.available() == size) {
        for (byte i = 0; i < size; i++) {
          data <<= 8;
          data |= Wire.read();                // receive DATA
        }
        if (data == 0) {data = 1;}
      }
    }
    x++;
  } while (Wire.endTransmission(true) != 0 && x <= I2C_RETRY_COUNTER); // end transmission
  return data;
}
```

arendst, 2017-11-27:
@stefanbode this is not an error for the HTU as no test is being performed on the result from I2CRead8.

My HTU21D works as expected:
[image]

It might be a problem for ADS1115 using my I2C routines and I will look into that.

stefanbode, 2017-11-27:
Ok, I tested with the AD1115 and here I definitely get the 0 as a result (1 byte returned on port x48). I added some debug messages to be clear that this is the issue.

arendst, 2017-11-27:
OK. I will rewrite the I2CRead in case of a timeout.

arendst, 2017-11-27:
Looked at it and I see no problem for both HTU and ADS1115. The routine might return a valid 0 but in both cases this result is not relevant.

Regarding HTU it does not test for a zero value so this is not a problem.

For the ADS1115 the only test is in the code below:
```
  for (byte i = 0; i < sizeof(ads1115_addresses); i++) {
    ads1115_address = ads1115_addresses[i];
    if (I2cRead16(ads1115_address, ADS1115_REG_POINTER_CONVERT)) {
      Ads1115StartComparator(i, ADS1115_REG_CONFIG_MODE_CONTIN);
      ads1115_type = 1;
      break;
    }
  }
```
where it expects to receive a "1" if an ADS1115 is present as did the I2CDev code:
```
  for (byte i = 0; i < sizeof(ads1115_addresses); i++) {
    ads1115_address = ads1115_addresses[i];
    ADS1115 adc0(ads1115_address);
    if (adc0.testConnection()) {
      adc0.initialize();
```
where the testConection() is nothing more than:
```
bool ADS1115::testConnection() {
    return I2Cdev::readWord(devAddr, ADS1115_RA_CONVERSION, buffer) == 1;
}
```
So I see no cause for the mentioned problems here.

arendst, 2017-11-28:
@RobKuipers  I suppose you didn't disable support for HTU21 in user_config.h (USE_HTU) ...

RobKuipers, 2017-11-28:
@arendst Nope. Just the plain pre-compiled sonoff.bin from the repository. Upgraded from 4.x should that make any difference.

arendst, 2017-11-28:
Should not make a difference. Try to enable more logging (weblog 3). Op the webconsole and execte command ``restart 1`` and see if a message like
```
21:16:57 I2C: HTU21 found at 0x40
```
is coming along.

RobKuipers, 2017-11-28:
Theo, it works now!
I'm not sure what did it in the end: I was in the process of resetting the whole module, when you send me your message saying I should try 'restart 1'. 
After resetting the whole thing, connecting it to Wifi, configuring the module again, and issuing 'restart 1', it works!
Sorry for all the inconvenience and a big thanks for your help,
Rob

RobKuipers, 2017-11-28:
Oops, that was a little to soon maybe... I get erratic readings now. Half the time correct. Half the time zeros or wrong values:

HTU21 Temperature | 19.8°C
-- | --
-0.8%

(minus 0.8% humidity)
Can this be related?

arendst, 2017-11-28:
How did you hook it up to your device? What GPIOs? Long lines? I2C pull up resistors? Do you have a picture of your device?

RobKuipers, 2017-11-28:
Lines are ~15 cm long. Pullups resistors are on the device.
GPIO 1 SDA
GPIO 3 SCL

I've attached a picture: 
[image]

Console says: 
21:49:29 MQT: tele/sonoff/SENSOR = {"Time":"2017-11-28T21:49:29", "HTU21":{"Temperature":19.7, "Humidity":**-0.8**}, "TempUnit":"C"}

arendst, 2017-11-28:
You might also want to swap the two around (1 = SCL, 3 = SDA)

RobKuipers, 2017-11-28:
Seriallog 0 and swapped lines.
Console reads:
22:07:50 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
22:07:50 MQT: stat/sonoff/POWER = OFF
22:07:56 SHT: Sensor did not ACK command
22:07:56 I2C: HTU21 found at 0x40
22:07:57 HTP: Main Menu
22:07:57 MQT: tele/sonoff/STATE = {"Time":"2017-11-28T22:07:57", "Uptime":0, "Vcc":3.186, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"H", "RSSI":44, "APMac":"C0:25:06:5A:CD:42"}}
22:07:57 MQT: tele/sonoff/SENSOR = {"Time":"2017-11-28T22:07:57", "HTU21":{"Temperature":19.4, "Humidity":54.2}, "TempUnit":"C"}
22:08:05 HTP: Console

The webpage still show erratic numbers from time to time. However I don't see these wrong readings in the console.

arendst, 2017-11-28:
Well at least it looks like your sensor is OK. 

I'm using 1 = SDA and 3 = SCL to drive a BMP180 and that runs fine so why the HTU is showing some erratic numbers I don't know.

RobKuipers, 2017-11-28:
Just so you know: I've just changed SCL to GPIO14 and still have the same results.

arendst, 2017-11-28:
Or your sensor is not OK....

RobKuipers, 2017-11-28:
I will order another one to make sure and meanwhile shorten the wires to see if that helps any.
Will get back with results.

Thanks again Theo!

RobKuipers, 2018-04-24:
Sorry I forgot to follow up on this issue. I did order a new sensor. But alas to no avail. The problem persisted.
Since apparently I'm the only one suffering from this issue, closing it might be best.

Thanks, Rob

closed 2018-04-26 by arendst

fernandorocha, 2020-02-18:
Same behaviour here. But mine doesn't show the `HTU21 found at 0x40`
```
20:36:00 CMD: I2CScan
20:36:00 SRC: WebConsole from 192.168.0.11
20:36:00 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
20:36:00 MQT: stat/devkit_F4B6FD/RESULT = {"I2CScan":"Device(s) found at 0x40"}
20:37:26 CMD: i2cdriver
20:37:26 SRC: WebConsole from 192.168.0.11
20:37:26 CMD: Group 0, Index 1, Command "I2CDRIVER", Data ""
20:37:26 MQT: stat/devkit_F4B6FD/RESULT = {"I2CDriver":"!7,!8,9,!10,!11,!15,!20"}

```

ljan, 2020-05-25:
Similar problem here but with the bme260

`15:55:34 CMD: i2cscan
15:55:35 MQT: stat/laternen/RESULT = {"I2CScan":"Device(s) found at 0x76"}
15:55:47 CMD: i2cdriver
15:55:47 MQT: stat/laternen/RESULT = {"I2CDriver":"7"}`