arendst/Tasmota #1260: Wemos D1 mini, I can't make it work with Tasmota

asimmetrico, 2017-11-27:
Hello,
I made some very successful test using IEAD Sonoff device. I want to try Tasmota on some Wemos D1 mini that currently work with ESPeasy.
I tried loading the standard procedure using Arduino IDE. After reset they don't work.
I looked for some previou topics. I found one that suggest to define some value to CFG_HOLDER to some arbitrary value. It is already set to 0x20161209.

How should I do to make Wemos D1 mini with Tasmota?

Thanks for your support

Alberto

smulle48, 2017-11-27:
Try to start the serial monitor to see what's going on, on the wemos d1

asimmetrico, 2017-11-27:
Is it there a tutorial on how to use the serial monitor?

smulle48, 2017-11-27:
I'm on Mac, but the menus should be the same on a windows machine.

With your wemos connected to your pc, select the menu "Tools" -> "Serial Monitor"

Now a new windows should pop up with some debug information from the wemos

asimmetrico, 2017-11-28:
I found on Arduino IDE. Quite easy if you know it :)

This is the output

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
⸮
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_4EDCBB, GroupTopic sonoffs) Version 5.8.0
00:00:00 WIF: Connecting to AP2 indebuurt2 in mode 11N as sonoff-7355...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 indebuurt1 in mode 11N as sonoff-7355...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting

asimmetrico, 2017-11-28:
I was able to connect to my AP harcoding the ssid and password in the config file. I'm not able to make start the internal AP.
I also tried to connect to ground GPIO0 4 times to simulate the sonoff button but without any result.
Is it there any solution?

Thanks

Alberto

stefanbode, 2017-11-28:
send though serial: wificonfig 2

WifiConfig | 2 | Start wifi manager (web server at 192.168.4.1) and set as current config tool
-- | -- | --

and enjoy the great documentation. There are hundreds of options.

[](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

davidelang, 2017-11-28:
without knowing what you configured, it's hard to say.

did you disable multipress to make the button more responsive for example?

If it is connecting to your AP, why do you need to have it start the internal 
AP? that's just a way to get access to it to make it connect to a real AP

asimmetrico, 2017-11-29:
@ Stefan: Thanks a lot. It works through serial monitor. Anyway after flashing the internal AP doesn't start.
@ davidelang: it is not configured. Just flashed. The AP doesn't start so I can't configure it unless I write in the code the config. It is useful to have it start the internal AP if it doesn't connect to any external AP becouse if you change the network (SSID for any reason) you don't have a blind ESP and you have to access it by HW. Imagine if you have 30 of them and you have to access all of them via serial monitor to change the SSID!

All the best

Alberto

stefanbode, 2017-11-29:
As far as I know it starts in WPS MODE. Therefore just activate WPS on our router and done. Even simpler than expected

asimmetrico, 2017-11-29:
I know. This is what happens with the Sonoff dev. When I flash the same on the Wemos it doesn't start in WPS mode.
Maybe some memory locations are not clean.
Is it there any way to overwrite all the 4Mb of the Wemos?

Thanks

Al

smulle48, 2017-11-29:
You Can try this guide to erase the entire memory

http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/

davidelang, 2017-11-29:
see the troubleshooting page on applying the config from the flashed image to 
the system

wifimode needs to be set appropriately to let you go into AP mode
k

davidelang, 2017-11-29:
see the troubleshooting page on the wiki, specifically the section on config 
problems

stefanbode, 2017-11-30:
There is an internal tasmoto command to do so: reset 2

tomgregg, 2018-02-21:
I used Arduino IDE to upload Sonoff-Tasmota, and it works great!  I put it on an ESP-12e module.   I don't see Wemos D1 mini in the list of Module Types.  Everything I see points to this Module Type as the first step to getting ADC support.  Do I need to comment out the USE_ADC_VCC statement in user_config.h first?  Do I need to switch from Arduino IDE to ESPeasy, PlatformIO or something else?  Thank you.