arendst/Tasmota #1261: Sonoff Brdige with Sonoff DW1

Odianosen25, 2017-11-27:
Hello @arendst ,

Please this is not an issue, but more like questions.

- I have just ordered a Sonoff bridge with a few DW1 Door/window sensors. I wish to confirm if after loading your firmware, I will be able to use the door sensors? 

- Are there extra settings I will need to do, that you recommend that I do before it works? 

- How many of such sensors can I have on a single bridge?

- What security should I put in place to ensure it all works fine.

Kind regards

ChukXP, 2017-11-27:
I've tested few days ago this firmware on Sonoff RF bridge. So I can confirm that when device send RF code, You've got mqtt message with this code from RF Bridge. So, I suppose that You can have any quantity of sensors.

Odianosen25, 2017-11-27:
Hello @ChukXP,

Oh thanks for the response, very much appreciated.

Kind regards

oriolism, 2018-01-30:
@ChukXP Do you receive an rf code when the sensor opens and another when the sensor is close or just one?

maartenpaauw, 2018-02-09:
@oriolism I connected my DW1 sensor to the Sonoff RF Bridge. I only receive a RF code when the sensor opens. Not when it close.

Anyone knows how to fix this?

Daemonentreiber, 2018-03-03:
@maartenpaauw This sensor doesnt send a signal when its closed.

charinduedirisuriya, 2018-04-21:
Is it possible to identify which sensor got triggered. Assume i attach one sensor to my front door and save it at key 1, and another sensor to my back door and save that as key2. Will tasmota be able to differentiate which door sensor sent the code, becouse both the sensors might send the same code. Does the sensors send same code or how exactly does the rf in the sensor work?

ascillato, 2018-06-09:
Hi @roger443 

How do you modify the DW to be normally closed?

closed 2018-06-27 by ascillato2

xFvPx, 2019-01-09:
@roger443 how to reverse the operation ?

pedrosimao, 2019-02-28:
I bought six DW1 sensors thinking it also signal when the window / door is closed, and I am quite disappointed. For me this is necessary because I want to know the current state of the window. As I live on the ground floor it would be easier for me than checking each individual window before leaving home.
Do you know another good alternative to Sonoff DW1 (maybe working with Sonoff bridge) that has both close and open signal?

ascillato, 2019-02-28:
Hi, keep searching in issues. This issue is super old. In another issue of DW1 there is a link to an alternative sensor.

daueee, 2019-07-05:
Do you now the alternative sensor?

ascillato, 2019-07-06:
Search for KERUI D026 window sensors

barriethesen, 2020-08-12:
On eWeLink the DW1 does not send out a notification until the sensor opens and then closes.  This is problematic because if an intruder just opens the door but does not close it there is no notification.  If the RF Bridge was Tasmotized I would think the same behaviour would repeat.