arendst/Tasmota #1262: Arilux LED-strip controller running Tasmota not working

speedyRon, 2017-11-27:
I am a happy Tasmota-sonoff user. Recently I redicovered the _Arilux AL-C01_ module to manage a led-strip. With the default software and a _magic-home_ android-app it was working fine but I want it to integrate with other devices using mqtt. So I installed Tasmota on the Arilux as described here [https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller](url). The soldering and flashing was done and seems ok. I can access the Tasmota installed webserver changed to module-type _"37 Arilux"_ and tried to send mqtt commands. Unfortunately I cannot light-up the arilux attached led-strip (nor change the settings). Most mqtt commands I find here [https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#ws2812-ailight-sonoff-led-b1-bn-sz01-h801-and-magichome](url) do not seem to work.

WebServer debug logging:
```00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on arilux1-7785.local with IP address 192.168.178.31
00:00:08 HTP: Main Menu
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tasmota/arilux1/tele/LWT = Online (retained)
00:00:08 MQT: tasmota/arilux1/cmnd/POWER = 
00:00:08 MQT: Subscribe to tasmota/arilux1/cmnd/#
00:00:08 MQT: Subscribe to tasmota/sonoffs/cmnd/#
00:00:08 MQT: Subscribe to cmnd/DVES_2A1E69/#
00:00:08 MQT: tasmota/arilux1/tele/INFO1 = {"Module":"Arilux", "Version":"5.9.1", "FallbackTopic":"DVES_2A1E69", "GroupTopic":"sonoffs"}
00:00:08 MQT: tasmota/arilux1/tele/INFO2 = {"WebServerMode":"Admin", "Hostname":"arilux1-7785", "IPAddress":"192.168.178.31"}
00:00:08 APP: (UTC) Mon Nov 27 17:21:05 2017
00:00:08 APP: (DST) Sun Mar 26 02:00:00 2017
00:00:08 APP: (STD) Sun Oct 29 03:00:00 2017
18:21:06 MQT: tasmota/arilux1/tele/INFO3 = {"RestartReason":"Software/System restart"}
18:21:07 MQT: tasmota/arilux1/stat/RESULT = {"POWER":"OFF"}
18:21:07 MQT: tasmota/arilux1/stat/POWER = OFF
18:21:07 HTP: Console
18:21:14 MQT: tasmota/arilux1/tele/STATE = {"Time":"2017-11-27T18:21:14", "Uptime":0, "Vcc":3.219, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"xxx", "RSSI":70, "APMac":"xxx"}}
18:22:44 WIF: Checking connection...
18:23:54 RSL: Received Topic tasmota/arilux1/cmnd/fade, Data Size 1, Data 1
18:23:54 RSL: Group 0, Index 1, Command FADE, Data 1
18:23:54 MQT: tasmota/arilux1/stat/RESULT = {"Command":"Unknown"}
18:24:05 WIF: Checking connection...
18:24:05 WIF: Connected
18:24:18 RSL: Received Topic tasmota/arilux1/cmnd/power, Data Size 2, Data ON
18:24:18 RSL: Group 0, Index 1, Command POWER, Data ON
18:24:18 MQT: tasmota/arilux1/stat/RESULT = {"POWER":"ON"}
18:24:18 MQT: tasmota/arilux1/stat/POWER = ON
18:24:19 CFG: Saved to flash at F7, Count 45, Bytes 1648
18:24:25 WIF: Checking connection...
18:24:25 WIF: Connected
```
When I reconfigure the module-type to _"36 MagicHome"_ I see the led on the Arilux module turn on when sending a _"tasmota/arilux1/cmnd/power ON"_, the LED-strip remains off.

What am I doing wrong, why does the led-strip not turn-on, why does the software not accept commands like _"Fade"_ or _"Color"_...?
Any help would be appriciated 😃

arendst, 2017-11-27:
Try command setoption15 1

speedyRon, 2017-11-27:
Thanks for the fast reponse, it works... great. 

Unfortunately the red channel on my arilux does not work anymore (no power on the pin), maybe it was damaged during the soldering...

Connecting the individual pins:
color ff0000 - green ok
color 00ff00 - (red) nothing
color 0000ff - blue ok

closed 2017-11-27 by speedyRon

johnsethwatkins, 2018-01-07:
I have just flashed 5.10.0 onto an Arilux LC-01, an "older" PCB one from the first half of 2017 with the pads on the bottom as per the wiki guide.

I cannot get any output on the red channel either. If I change the type to 34 MagicHome, I can get all 3 (RGB) channels working, albeit mixed up. With 37 Arilux the red channel is dead.

Is there an easy way to change the pin assignments in the sketch before compiling which would fix this?

mrohner, 2018-02-08:
Same here, no red color can be displayed on my Arilux LC01 with Tasmota. Blue and green ok.

<16:01:08 CMD: status 0
16:01:08 MQT: stat/arilux74/STATUS = {"Status":{"Module":37,"FriendlyName":"KitchenTop","Topic":"arilux74","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
16:01:08 MQT: stat/arilux74/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://127.0.0.1:8887/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":10,"SaveCount":35,"SaveAddress":"F9000"}}
16:01:08 MQT: stat/arilux74/STATUS2 = {"StatusFWR":{"Version":"5.11.1f","BuildDateTime":"2018-02-08T14:47:16","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
16:01:08 MQT: stat/arilux74/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"mmwireless","SSId2":"","TelePeriod":300,"SetOption":"00008009"}}
16:01:08 MQT: stat/arilux74/STATUS4 = {"StatusMEM":{"ProgramSize":451,"Free":552,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
16:01:08 MQT: stat/arilux74/STATUS5 = {"StatusNET":{"Hostname":"arilux74-7748","IPAddress":"192.168.178.74","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"A0:20:A6:0D:7E:44","Webserver":2,"WifiConfig":3}}
16:01:08 MQT: stat/arilux74/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.59","MqttPort":8883,"MqttClientMask":"DVES_X","MqttClient":"DVES_0D7E44","MqttUser":"mqtt","MAX_PACKET_SIZE":1024,"KEEPALIVE":15}}
16:01:08 MQT: stat/arilux74/STATUS7 = {"StatusTIM":{"UTC":"Thu Feb 08 15:01:08 2018","Local":"Thu Feb 08 16:01:08 2018","StartDST":"Sun Apr 01 02:00:00 2018","EndDST":"Sun Oct 28 05:00:00 2018","Timezone":99}}
16:01:08 MQT: stat/arilux74/STATUS10 = {"StatusSNS":{"Time":"2018-02-08T16:01:08"}}
16:01:08 MQT: stat/arilux74/STATUS11 = {"StatusSTS":{"Time":"2018-02-08T16:01:08","Uptime":0,"Vcc":3.171,"POWER":"OFF","Dimmer":100,"Color":"FF0000","Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"mmwireless","RSSI":60,"APMac":"0E:41:58:01:03:F0"}}}>

arendst, 2018-02-08:
So you're red channel doesn't work. Any idea what could be the problem here? Hint: it might be that your hardware is different from my LC01 where the red channel is connected to GPIO05.

So what do you think would be your next step....

1) get familiar with file sonoff_template.h
2) find the configuration for a Arilux LC01 around line 719
3) Study what you see here; Colors are defined connected to some GPIOs as PWM
4) Now on your LC01 hardware open the configuration webpage and select module Generic, wait for the restart to complete and change some GPIOs to match the config from step 3.
5) Make sure ``SetOption15 1`` is executed after the restart and verify the same erratic behaviour as before on your module Arilux LC01.
6) now for the fun part play around with PWM1 on another GPIO in the module config to find the correct GPIO to be used for color Red.
7) If found you're done and have to live with the Generic Module servicing your Arilux LC01
8) If not found study your hardware for a mis-wire or broken channel

mrohner, 2018-02-09:
Thanks arendst. After some tedious testing I found a PWM combination that worked for my Arilux LC01:
[image]