arendst/Tasmota #1263: Upgrade 5.8.0->5.9.1j. Missing all web info

Tresett, 2017-11-27:
Hi all.
I'm upload old fw (5.8.0) to new Sonoff Basic via serial port. Succesefful.
Then clone project via Github software to new folder, edit user_config.h, compile and upload via web.
Information page is empty. Device list is empty, console does not show nothing information, etc ...
Compile options: 40Mhz, DOUT, 1M (NoSpiff)
What I'm do wrong?

P.S. Old firmware normal uploaded and all information back to the pages.

Tresett, 2017-11-27:
Hm. Wrong.
At 5.8.0 I see full Information page, but in Configuration Module->list modules and other parameters is empty.

smulle48, 2017-11-27:
Try to upload the already compiled firmware to see if you still got the same issues.

https://github.com/arendst/Sonoff-Tasmota/releases.

The son off with start in AP mode so you can connect directly to it by wifi and configure your network

Tresett, 2017-11-27:
Same-same.
Upload already compiled FW, restart, press 4 touch, connect to AP, configure WiFiNetwork.
Restart ... and Information page is empty, module list is empty.

P.S. My order contains 5 basic (try 2), 1 Pow, 1 RF Bridge. Other hw for tomorrow, not now.

smulle48, 2017-11-27:
Have you tried a different browser? :-D

What does your serial monitor output?

Tresett, 2017-11-28:
Bingo!
FF plugin "NoScript" block JS
=)

Tnx to all!

closed 2017-11-28 by Tresett