arendst/Tasmota #1265: WeMos D1 mini, wrong behavior - code condition for button

roblad, 2017-11-28:
Hi,
WeMos D1 mini -  behavior.

Problem with LED and button without defined relay for 1 GPIO. I have found that if no GPIO is defined for 1 relay, it is not reported MQTT button state nor the web has no button, also does not respond to defined LED on GPIO2 - no action, module does not respond to defined button on GPIO0, button commands do not work. After defining the relay for any GPIO, everything works fine. Haw to see button behaviour on MQTT OFF/ON and see status on www when Relay is not defined,  and haw to set button on GPIO0 for proper behaviour when no relay defined?

stefanbode, 2017-11-28:
Yes, the BUTTON is strongly connected to RELAY functions. The other multipress ones you only get with a relay. Maybe it makes sense to remove the relation to the relay and allow the multipress commands also without a relay defined. If not a major problem just assign the relays to a spare GPIO like the 0

roblad, 2017-11-28:
Hi,

Hmmm, on GPIO0 I have Button1 - haw to set -"remove the relation to the relay and allow the multipress commands also without a relay defined" Where it is in config or wchich Set... I need to switch it on, or may be there is some change in the code required. I need this functionality to monitor switch on/off - there will be some connector for contrlol state of whater level. I need report it through MQTT, I do not need relay but button I need. Additionaly I am not able to operate on spetial button functions i.e. AP mode , restart etc. If you can do any option in config to switch off relations and use button without relays it will be usefull for such functionality as I want. I want to save GPIO for other sensors even 1 spare GPIO only for enabling the button functionality does not make sense.

stefanbode, 2017-11-28:
Why don't you use the SWITCH1 in TOGGLE mode. Then you have the same functionality ON/OF and can switch the LED1 without having a relay defined. you can reach me on Skype if interested to discuss

roblad, 2017-11-28:
may be hangauts ? give me your e-mail on google I do not use skype

closed 2018-05-07 by stale[bot]