arendst/Tasmota #1266: getting repeating "Command: Unknown"

rhodopsik, 2017-11-28:
I'm using v5.9.1i running at Nodemcu, 
controlling a SK6812 (GRBW-) LED-String (comparable to ws2812b) (USE_WS2812_CTYPE = 3).

It works fine, until the moment I try to set a specific color with following command at the console:
```
10:24:03 CMD: color #000000FF
10:24:03 MQT: stat/LED/RESULT = {"POWER":"ON","Dimmer":100,"Color":"000000FF"}
```
after that, I'm getting repeating such messages, until the Nodemcu crashes or initiates a restart:

```
10:25:24 CMD: L
10:25:24 MQT: stat/LED/RESULT = {"Command":"Unknown"}
10:25:39 CMD: @@@m
10:25:39 MQT: stat/LED/RESULT = {"Command":"Unknown"}
10:25:40 CMD: 
10:25:40 MQT: stat/LED/RESULT = {"Command":"Unknown"}
10:26:32 CMD: P@
10:26:32 MQT: stat/LED/RESULT = {"Command":"Unknown"}
10:26:36 CMD: H6
10:26:36 MQT: stat/LED/RESULT = {"Command":"Unknown"}
10:27:30 CMD: L@`E
10:27:30 MQT: stat/LED/RESULT = {"Command":"Unknown"}
10:28:05 CMD: 6HPhI@@PI
10:28:05 MQT: stat/LED/RESULT = {"Command":"Unknown"}
```
Does anybody know about the reason for that behaviour?

stefanbode, 2017-11-28:
I also had this one time and it was a defect WEMOS mini, that might have seen a bit too much moisture.

davidelang, 2017-11-28:
any chance that you defined a serial port as being used by something else?

This is telling you that you are getting strange commands, and they are commands 
that don't make sense, so the question is how these commands are getting to the 
device. The most likely way is through the serial port if something is wired 
wrong.

arendst, 2017-11-28:
If you power your led string from the Nodemcu expect serious problems as the power supply of the Nodemcu is unable to supply enough power for the leds. This will lower the Nodemcu power and anything could happen.

BTW did a test (without the leds as I do not own a SK6812) and did not experience your experience.

rhodopsik, 2017-11-28:
Thanks for replying!
After reading your comments, I disconnected the Nodemcu for a while and gave it a new try. And what can I say: Until now there is now "ghost command" occuring. Perhaps there was some kind of random elektro magnetical problem or somewhat...
Now it seems to be working proper...
And no, Theo, I'm not driving my 8 meters Strip with 486 LEDs by the nodemcu's supply ;-).
But thanks for your advice, I also will never try that ;-)

stefanbode, 2017-11-29:
Please close, if solved

closed 2017-11-29 by rhodopsik