arendst/Tasmota #1267: ADS1115 with 5.9.1j only partially fixed

stefanbode, 2017-11-28:
hi @arendst,

with the newest build the detection of the ADS1115 is back working; thanks. But I get for all 4 channels a 0, 0 from the I2Cread16. Try to get an idea whats wrong but did not get it so far. Maybe a mistake in the config write? to get the values, because I cannot find an issue in the read procedure.

stefanbode, 2017-11-28:
yes the loop definition is wrong. It should be following. now working

```
bool I2cWrite(uint8_t addr, uint8_t reg, uint32_t val, uint8_t size)
{
  byte x = I2C_RETRY_COUNTER;

  do {
    Wire.beginTransmission((uint8_t)addr);              // start transmission to device
    Wire.write(reg);                                    // sends register address to write to
    uint8_t loops = size ;
    do {
      Wire.write((val >> (8 * (loops-1))) & 0xFF);          // write data
    } while(--loops);
    x--;
  } while (Wire.endTransmission(true) != 0 && x != 0);  // end transmission
  return (x);
}
```

arendst, 2017-11-28:
Should work fine now. Pls test

stefanbode, 2017-11-29:
I‘m OOTO the next two days. Have no ADS1115 to test with me. Because ‚size‘ is a local variable you can use this directly. ‚Byte‘ as an additional variable is not required in my mind. Additionally I learned that here is a difference in ––loop and loop–– if used in an if clause. Interesting. The test is done before or after changing the variable.

stefanbode, 2017-12-01:
OK, works fine for me. Also the version without "byte" variable.

```
bool I2cWrite(uint8_t addr, uint8_t reg, uint32_t val, uint8_t size)
{
  byte x = I2C_RETRY_COUNTER;
  do {
    Wire.beginTransmission((uint8_t)addr);              // start transmission to device
    Wire.write(reg);                                    // sends register address to write to
    while (size--) {
      Wire.write((val >> (8 * size)) & 0xFF);          // write data
    }
    x--;
  } while (Wire.endTransmission(true) != 0 && x != 0);  // end transmission
  return (x);
}
```

closed 2017-12-01 by stefanbode

BIP-BUD, 2020-09-08:
Hi, maybe I'm running now in to the same or similar issue with v8.5.0. All my 4 channels are showing 0. The ADS1115 will be recognized (I2Cscan) and show up in the webUI. I tried with tasmota-sensors.bin and also with my own compiled versions from Tasmota-master and also Tasmota-development. Unfortunately without luck. Can you give me a hint what to check (I'm quite new to Tasmota)?

ascillato, 2020-09-08:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

BIP-BUD, 2020-09-08:
Sorry, I thought as I have the same symptoms but with a newer version it will fit here. 
Tx for guiding me to the right place.