arendst/Tasmota #1268: DS18B20 new JSON format is pain on parsing because of "-" in identifier

stefanbode, 2017-11-28:
Hi,

I may not know, but parsing a "-" in an identifier does not work really well. It will be recognized as an operator.
This is not working:
`JSON.parse(input).DS18B20-3.Temperature;`

if we can replace with a _ everything would be fine:
`JSON.parse(input).DS18B20_3.Temperature;`

davidelang, 2017-11-28:
you need something to tell your code that DS18B20-3 is one entity. I don't know 
what language you are using, but putting {} around them or using quotes is a 
common method

arendst, 2017-11-28:
From www.json.org:

[image]

Problem regarding the dash have been solved by people using more DHT's or AM2301's like:
```
"AM2301-12":{"Temperature":22.3,"Humidity":48.0},"AM2301-14":{"Temperature":21.8,"Humidity":47.5}
```
Search around here for a solution.

stefanbode, 2017-11-28:
As you can see Ivuse plain Javascript. This does not like the -   Also putting “ around did not help. Will try with { but assume this is also a problem. Also tried \ to mask the -. Also no success. Any idea, how the JavaScript should look like.

arendst, 2017-11-28:
Google is your friend https://stackoverflow.com/questions/13869627/unable-to-access-json-property-with-dash

AlexSwensen, 2017-11-29:
This is not an issue with the firmware, but an issue with understanding its api.

This issue should be closed.

closed 2017-11-29 by stefanbode