arendst/Tasmota #1269: MAX31850 temp is 20 degrees low at room temp

Beef-well, 2017-11-28:
I'm using a MAX31850 to monitor my wife's kiln. Before I updated to 5.9.1 it seemed to be much closer to the other temp sensors I've got. Now it's 20 degrees F lower than the other sensors at room temperature and it goes to -7000 when heat is applied.

Beef-well, 2017-11-28:
I also have a DS18B20 connected to the same SonOff on GPIO4 and it's fine. I have both set to DS18x20 I believe.

arendst, 2017-11-28:
Tasmota only supports ONE gpio per device to be used for OneWire. So either your DS18B20 to one GPIO or your MAX31850 to the other GPIO.

Why not connect both to the same GPIO. Using the multi sensor driver (Enabled in user_config.h and with the lates development version you now even have a choice of two)

Beef-well, 2017-11-28:
I'll try that. It hadn't even crossed my mind to put them on the same GPIO.

Beef-well, 2017-12-03:
It turns out that I did have both sensors connected to GPIO 14. It’s been awhile since I originally set this up. However the MAX31850 is consistently ~24F degrees lower than the ds18b20. Also after the MAX31850 reaches 265F it jumps to around -6500F. I’ve been looking at Ds18b20.ino, but I’m not sure what to change to adjust the temp and even if I adjusted the temp I’m not sure what that will do to the reading going to -6500. Any guidance would be much appreciated.

arendst, 2017-12-03:
The max is only supported by the ds18x20.ino and ds18x20_legacy.ino files.

If debugging max try to connect the max only using one of above two driver files selectable in user_config.h

simonjcoates, 2018-03-08:
A bit of an old thread I know but since I had a similar issue to Beef-well that lead me here I thought I would post my findings.

It looks like the MAX31850 code in ds18x20.ino is still handling values from the sensor as though it was a DS18B20. For the DS18B20 these are:

S S S S S 2^6 2^5 2^4 (hi byte)
2^3 2^2 2^1 2^0 2^-1 2^-2 2^-3 2^-4 (lo byte)

Treating these as an unsigned int will yield positive values <=2047 and negative values above which is what the code does at lines 329 and 320. However, the MAX31850 represents temperature values a bit differently:

S 2^10 2^9 2^8 2^7 2^6 2^5 2^4 (hi byte)
2^3 2^2 2^1 2^0 2^-1 2^-2 X X (lo byte)

So anything up to and including 32767 should be treated as a positive value. This is why readings up to 127C (roughly 260F) seem OK but then suddenly jump negative. The fix is simple enough. Either modify the test value at line 320 to 32767 or treat the value as signed and test for < 0. I did the latter and it seems to be working (although I have no way to easily test for negative values from the sensor given that it is in a boiler flue).

      case MAX31850_CHIPID:
        int16_t temp12 = (data[1] << 8) + data[0]; // Was uint16_t
        if (temp12 < 0) { // Was temp12 > 2047
          temp12 = (~temp12) +1;
          sign = -1;
        }