arendst/Tasmota #1272: Help flashing Wifi LED Strip

tatunts, 2017-11-28:
Hello everyone!

I've bout a few Wifi LED controllers and they arrived today. I was following the Wiki's instructions but got very confused about the board I received. It is very different from the pictures provided here.

Here are some photos of my controller: https://photos.app.goo.gl/0C2owhZdNrthjw2s2

Can someone please help me?

Thanks a lot!

arendst, 2017-11-28:
They look like AriLux AR-LC01 to me. I kind of did it the following way:

To not solder anything on it I propose to use a USB to serial board with pogopins connected to Gnd, GPIO0, Tx and Rx directly on the ESP8266 chip.

Just google for the correct connections. Once the connections are made (especially GPIO0 to Gnd) add power to the unit from an external power supply and start flashing.

tatunts, 2017-11-28:
Wow! Thank you so much @arendst for taking the time to answer my issue!

I've look to this board for almost an hour now and I could not find the RX and TX connections! hahaha

Can you please guide me to information on where to find it?

tatunts, 2017-11-28:
This are my available tools: https://photos.google.com/share/AF1QipM83AGgLR_BWhklY1cv1TRgk8KHS3es1336MZfMdM4c880NUjJp_UcgEniJ-qmPAA/photo/AF1QipMTSPc8Pqht0tgkfXjRcM6e98VB5QAuwSC0hNuD?key=VHJIY1ZjX3g4dF9NTmtEMEl0VUxERjNScjdXX0h3

vcordes79, 2017-12-06:
Got some of these too. I found this page, looks like it's the one:
https://github.com/SmartArduino/SZDOITWiKi/wiki/ESP8285---ESP-M2
There's a link to the manual

tatunts, 2017-12-06:
Hello @vcordes79, I found some information here: https://bitbucket.org/xoseperez/espurna/issues/231/updated-wiring-instructions-for-magic-led

I was able to successfully flash with TASMOTA!

tatunts, 2017-12-12:
hello @eggfat! No problem. I've soldered these 4 cables and touched the GPIO0 on the ground pin when connecting the 12v power. After a few seconds, separate the gpio cable from GND pin on the USB flashing tool and start flashing it. This is how I did it.

eggfat, 2017-12-13:
Hi @tatunts. Thanks for the reply.
I have rewired the boards and followed the instructions.
I think its in flash mode. PlatformIO says its uploading .pioenvs/sonoff/firmware.bin and the orange RX is flashing on the USB flashing tool.
Its been uploading for 30 minuets. 
Did yours take so long to flash?

Many thanks.

tatunts, 2017-12-13:
Hello again! I'm sorry that you were still not able to flash it... This is very odd indeed. Actually, I use the Arduino IDE software to flash those chips. I have a portable version on a folder all set up to flash the sonoff basic module. I've changed the project name and start flashing the LED module right away. It takes about 30 seconds to flash. The lights blinks and stops on the USB tool, then I unplug the DC and finally the boards cables. I am impressed that I was able to do it because I have never done anything like this before. There are many variables and things to go wrong hehe Does the blinking lights on your board never stops when flashing?

eggfat, 2017-12-13:
Hi there. Yes the RX light constantly flashes. I disconnected an reconnected and the same thing happened. I must have wired wrong. I will try again later. Thanks for your help.

tatunts, 2017-12-13:
That's weird... it looks like it is in flash mode and receiving the data, like you said. Have you tried using the Arduino IDE? Sounds like your settings are not right for this chip...

rathgar, 2018-01-27:
Looks like they are using that board now - I got some too. I tried to follow @tatunts link but the issue no longer exists. One odd thing I did notice is that when powered it is open on `:80` but doesn't seem to respond with anything useful. I have tried SonOTA but in vain.
[image]
Anyone have any help on this this board, preferably solderless.

10mabo, 2018-02-06:
I also have this board. Got it flashed with cables directly soldered to the chip :-)

[image]