arendst/Tasmota #1274: SwitchMode and SwitchTopic Confusion

GlennToms, 2017-11-29:
I've done a fair bit of research and I don't think its possible but I need to ask.

Sonoff Basic:
Is it possible to have one GPIO toggle relays directly and have another GPIO send MQTT messages about HIGH/LOW state?
I want one Hardware Light Switch so if MQTT is down lights still work and a PIR sensor to send MQTT but not change relay/light.

If the above isn't possible is it development or is there a fork I can join?

mkh595, 2017-11-30:
It's possible.

External button on GPIO3, PIR sensor on GPIO1, backlight of button on GPIO14. I used PWM as I can control Brightness of backlight. PIR sends status with topic defined in SwitchTopic.

In my case it can be switched manually via button or from HA (topic cmnd/home208/POWER) or automaticall via pir between sunset and sunrise (night). During day only backlight as indicator that MQTT is working. Control (backlight, light On/Off and timeout) is done in Node Red. Normally 5 seconds of backlight, only when light is switched off, backlight is turn on for 30 seconds.

[image]

[image]

```
10:25:30 MQT: cmnd/home208pir/POWER1 = ON
10:25:30 MQT: stat/home208/RESULT = {"PWM":{"PWM1":500}}
10:25:35 MQT: cmnd/home208pir/POWER1 = OFF
10:25:35 MQT: stat/home208/RESULT = {"PWM":{"PWM1":0}}
10:25:39 MQT: cmnd/home208pir/POWER1 = ON
10:25:39 MQT: stat/home208/RESULT = {"PWM":{"PWM1":500}}
10:25:42 MQT: cmnd/home208pir/POWER1 = OFF
10:25:44 MQT: stat/home208/RESULT = {"PWM":{"PWM1":0}}
10:30:33 CMD: cmnd/SwitchMode
10:30:33 MQT: stat/home208/RESULT = {"SwitchMode1":1}
10:30:52 CMD: cmnd/ButtonTopic
10:30:52 MQT: stat/home208/RESULT = {"ButtonTopic":""}
10:30:59 CMD: cmnd/SwitchTopic
10:30:59 MQT: stat/home208/RESULT = {"SwitchTopic":"home208pir"}

GlennToms, 2017-11-30:
How did you get "13 Button1" to show up?  Its not listed on mine I'm on Version 5.9.1.

PS, I love all the extras you've added. PWM light indicator is awesome!

mkh595, 2017-11-30:
I changed GPIO0 default settings to GPIO_USER which allows Button 1 to be used on any pin. Then I can have all connections via 5 pin header. I don't want to solder extra cable from external button to built in button on GPIO0.

[image]

GlennToms, 2017-12-02:
Unfortunately I have a switch and you have a button.  This means if i turn on my switch the button is held on and device resets.

Is there any way to send MQTT messages about a single GPIO State change but leave the SwitchMode and SwitchTopics as default?

davidelang, 2017-12-02:
On Sat, 2 Dec 2017, Glenn Toms wrote:

> Is there any way to send MQTT messages about a single GPIO State change but 
> leave the SwitchMode and SwitchTopics as default?

no, the nly way to detect GPIO state changes is with switches and buttons. 
buttons are momentary switches designed to toggle relays

the only way to have them send messages by themselrves is to set buttontopic or 
switch topic (depending on which you are using)

switchmode is needed to define what transitions of the GPIO are reported and how 
they are reported

GlennToms, 2017-12-02:
Yep that's what my research pointing too, thanks for informing me. I'm working on fork that will add Generic Sensors that report state changes over MQTT.

closed 2017-12-02 by GlennToms