arendst/Tasmota #1275: Webfrontend: Add footer with the usual info

ThomDietrich, 2017-11-29:
I just realized that the webfrontend doesn't actually contain any product information. No GitHub link, no author details, the page doesn't even mention "Tasmota" once. Is that intentional? 

I propose to add a static footer with "Sonoff-Tasmota", the installed firmware version and the GitHub Repo Link.

The same goes for the page title. Currently it states "Main Menu" or similar. Pretty useless with multiple tabs open. Consider "Main Menu - ${Friendly Name} - Sonoff Tasmota Module"

arendst, 2017-11-29:
Great idea at the cost of about 250 bytes:

[image]

lgnd33, 2017-11-29:
or additional 41 for "<a href="http://bit.ly/2ndbFAt">GitHub</a>"

you should probably register the shortURL, to make it permanent.

arendst, 2017-11-29:
As always there is more going on beneath the waves ;-)
```
<div style='text-align:right;font-size:11px;'><hr/><a href='https://github.com/arendst/Sonoff-Tasmota' target='_blank' style='color:#aaa;'>Sonoff-Tasmota {a} by Theo Arends</a></div>
 ```
Where {a} is being replaced by the version number.

I think I have to settle with that...

lgnd33, 2017-11-29:
Do a URL shortener - that should save few bytes here and there.

ThomDietrich, 2017-11-29:
LGTM. Thanks!

closed 2017-11-29 by ThomDietrich