arendst/Tasmota #1277: platformio error on build - "multiple definition"

donster22, 2017-11-29:
hi,

i'm trying to build sonoff-tasmota in platformio before uploading it to a magichome led controller.
each time i try to build it i get an error message (copied below). I've been through all of the issues on github but cannot find this one so any help would be greatly appreciated. I'm on windows 10 btw.

    .pioenvs\sonoff\src\main.o:main.cpp:(.text.setup+0x0): first defined here
    .pioenvs\sonoff\src\sonoff.ino.o: In function `loop':
    sonoff.ino.cpp:(.text.loop+0x3c): multiple definition of `loop'
    .pioenvs\sonoff\src\main.o:main.cpp:(.text.loop+0x0): first defined here
    collect2.exe: error: ld returned 1 exit status
    *** [.pioenvs\sonoff\firmware.elf] Error 1
    [ERROR] Took 110.33 seconds
 
    [SUMMARY]
    Environment sonoff-NL           [SKIP]
    Environment sonoff-DE           [SKIP]
    Environment sonoff-PL           [SKIP]
    Environment sonoff-minimal      [SKIP]
    Environment sonoff-ds18x20      [SKIP]
    Environment esp12e              [SKIP]
    Environment sonoff              [ERROR]
    [ERROR] Took 110.33 seconds
 
Thanks for any assistance.

joecotton, 2017-11-29:
How did you obtain the source tree? There should not be a sonoff/src/main.cpp file
Did you perhaps create a new project in PlatformIO and then copy the git tree into it?

donster22, 2017-11-29:
To be sure, i deleted my projects and re-downloaded the zip file. i then unpacked it to my desktop and clicked "new project" in platformio and chose the unpacked folder on the desktop. now i'm getting a similar error on compile.

    Linking .pioenvs\sonoff\firmware.elf
    .pioenvs\sonoff\src\sonoff.ino.o: In function `setup':
    sonoff.ino.cpp:(.text.setup+0x108): multiple definition of `setup'
    .pioenvs\sonoff\src\main.o:main.cpp:(.text.setup+0x0): first defined here
    .pioenvs\sonoff\src\sonoff.ino.o: In function `loop':
    sonoff.ino.cpp:(.text.loop+0x3c): multiple definition of `loop'
    .pioenvs\sonoff\src\main.o:main.cpp:(.text.loop+0x0): first defined here
    collect2.exe: error: ld returned 1 exit status
    *** [.pioenvs\sonoff\firmware.elf] Error 1
    [ERROR] Took 64.40 seconds
     
    [SUMMARY]
    Environment sonoff-NL           [SKIP]
    Environment sonoff-DE           [SKIP]
    Environment sonoff-PL           [SKIP]
    Environment sonoff-minimal      [SKIP]
    Environment sonoff-ds18x20      [SKIP]
    Environment sonoff              [ERROR]
    [ERROR] Took 64.40 seconds

joecotton, 2017-11-29:
You should not need to do the "new project" function. The source release zip is already a platformIO project. You can just open the directory in PlatformIO/Atom and go from there.

donster22, 2017-11-29:
thanks. that helped. sorry for my basic lack of knowledge. i've managed to upload it all now. i can't find it on the network yet but I'm a step further on. thank you. i'll go off and read some more. don

closed 2018-05-07 by stale[bot]