arendst/Tasmota #1283: Pre-configure sonoff TH10?

SrKlein86, 2017-11-30:
Hi all. I'm in charge of flashing and configuring 30+ sonoff TH10, and I'd like to know if there's a way to configure what kind of sensor and the GPIO pins it uses before compiling, since it would save me a lot of time , having to configure each device separately.
Thanks in advance

KPK75, 2017-11-30:
Configure one. Backup config and restore to the others

SrKlein86, 2017-12-01:
This way I would still need to flash and configure each sensor, right? I mean, I'd have to log into each device and restore the backup individually, so it would take the same time/effort as a manual configuration. Or am I missing something? maybe this config backup can be flashed instead?

KPK75, 2017-12-01:
Just flash and then only restore and all should be setup. Or prepare script to configure them via mqtt.

davidelang, 2017-12-01:
remember, you can set all this via mqtt as well. As long as our devices connect 
to the mqtt server you can automate the rest.

Frogmore42, 2017-12-03:
You are going to have to configure each one to get them to have unique names, so it will require some patience no matter what.  But, to do what you asked, you would need to modify two files:
in sonoff.h
`#define MODULE                 SONOFF_BASIC // [Module] Select default model
`
change SONOFF_BASIC to the module you want SONOFF_TH

And then in sonoff_template.h modify the definition of SONOFF_TH in
`const mytmplt kModules[MAXMODULE] PROGMEM = {`

to be configured the way you want.

Note, that doing this will make it harder to integrate future changes to the core SW.

SrKlein86, 2017-12-04:
thank you so much for your answers! I'd rather tinker with a couple lines of code before flashing each device than configure/restore each one of them individually.
@Frogmore42 
with _unique names_ you mean the one configured with "PROJECT" macro in user_config.h ?
Also, you specified how to change to SONOFF_TH, but is there a way to pre-configure the sensor type and GPIO pin?
Thanks in advance

Frogmore42, 2017-12-04:
`#define MQTT_TOPIC             PROJECT           // [Topic] (unique) MQTT device topic`

`  #define FRIENDLY_NAME        "Sonoff"          // [FriendlyName] Friendlyname up to 32 characters used by webpages and Alexa`

Yes, you definitively need to set up MQTT_TOPIC to be unique if you want to sue MQTT (and if you don't why are you using Tasmota :-).  You should also set up FRIENDLY_NAME to be unique, but I believe you can get away with having it not be unique in some/many cases.

SrKlein86, 2017-12-04:
Yes, indeed. When I first tried tasmota (v3 i think), I only configured the project macro and it worked perfectly. Anyway, now that you mention it, it wont hurt to keep the friendly name unique too.

closed 2017-12-04 by SrKlein86