arendst/Tasmota #1285: Sonoff POW Domoticz

tobang, 2017-11-30:
I am no able to retrieve the power usage in Domoticz. I am using Electric (Instant + Counter), but no data is sent to Domoticz.

Current, Voltage and Power are working fine.

On the Sonoff i set the idx on Sensor idx 6 Count

Torben

arendst, 2017-11-30:
Tasmota sends this to Domoticz:
```
      DomoticzSensorPowerEnergy((uint16_t)watts, stotal_energy);  // PowerUsage, EnergyToday
      DomoticzSensor(DZ_VOLTAGE, svoltage);  // Voltage
      DomoticzSensor(DZ_CURRENT, scurrent);  // Current
```
and my Domoticz config looks like:

[image]

Resulting in this:

[image]

tobang, 2017-11-30:
Thank you, got it working looking at your screen dumps. 

Thank you,
Torben

closed 2017-11-30 by tobang