arendst/Tasmota #1286: DST for Israel

issacg, 2017-11-30:
Hi,

I'm trying to figure out how to set TIME_DST for Israel.

Israel has very odd timing for DST.  [According to Wikipedia](https://en.wikipedia.org/wiki/Israel_Summer_Time#2013.E2.80.93Present), the current definition is the Friday before the last Sunday in March at 02:00

Any pointers as how to set that in TIME_DST?

arendst, 2017-12-02:
You must be able to make the change in user_config.h and re-compile.

issacg, 2017-12-02:
Yes, I already understood that.  I'm having trouble finding what powers the DSL for the summer-time rules.  I can make it the last sunday in march at 02:00 really easily, but my question is how would I make it the *Friday before* the last Sunday in March?

arendst, 2017-12-02:
Ah. You can't. Time to move ;-)

arendst, 2017-12-02:
I understand the problem and for now the only solution I see is to configure either the last friday OR the last sunday in March. During these few days your time will be off but after a week it should be fine again as it must be DST time then.

I expect major coding additions to tackle this anomaly (as I see it). Sorry.

arendst, 2017-12-02:
From this table

[image]

You could choose to set DST for the next year:
```
#define TIME_DST               North, Fourth, Fri, Mar, 2, +180 
```
and after that for the next 5 year set it to:
```
#define TIME_DST               North, Last, Fri, Mar, 2, +180 
```

issacg, 2017-12-03:
Thanks.  Just wanted to make sure I wasn't just unaware of some cool macros.

You're right that it's anomoly - summer time rules have always been messed up here ;)

closed 2017-12-03 by issacg