arendst/Tasmota #1287: Power On for X seconds via http?

JoeWiseman, 2017-12-01:
Hi there,

I'm using my sonoff devices along with the homematic-CCU which does not speak any MQTT at all, so I'm using the devices by controlling them via http-commands.

Now I would like to know if there is a command to power on a device for a certain time like the light in stairways ... Once you press the switch the light is turned on and it is automatically turned of after X seconds?

Acutually I'm doing this by using two scripts where the first one is switching the light on and the second one - delayed after X seconds - is switching the light off. 

Would be great if anyone can tell me if there is such a feature....

Regards

Joe

lgnd33, 2017-12-01:
Take a look at the backlog command.

KPK75, 2017-12-01:
Or pulsetime command

jetema, 2017-12-01:
Have a look here --> https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

PulseTime<x> |   | Show current PulseTime of relay<x> in 0.1 seconds
PulseTime<x> | 0 / off | (Default) Disable use of PulseTime for relay<x>
PulseTime<x> | 1..111 | Set PulseTime for relay<x> with 0.1 seconds increment
PulseTime<x> | 112..64900 | Set PulseTime for relay<x> with 1 seconds increment starting with 12 seconds (113 = 13 seconds etc.)

JoeWiseman, 2017-12-03:
Hi everybody!
thanks a lot for your helpful answers! It wourked out great!!
Sometimes it is hard to see the wood because of the trees ;-)

closed 2018-05-07 by stale[bot]