arendst/Tasmota #1288: "Connecting" sonoff-devices by http-commands

JoeWiseman, 2017-12-01:
Hi there,

I'm using my sonoff devices along with the homematic-CCU which does not speak any MQTT at all, so I'm using the devices by controlling them via http-commands.

I have a Sonos-sound system in my living room. Though they can't be shut down I like to cut off / switch on the power by using some Sonoff-devices. There are three devices needed, which are distributed in the room... 

Now I thought it would be great, if I press switch on one device that there is a http-command send out from the device to switch on the other two devices ... And of course if I switch on one from the other devices that they are starting the left devices as well ...  Just like an "After Power On"-Command.

The problem - when using the http-comands - is, that you don't get an active feedback. It is possible that I get the status (on/off) by http-command. But I have to run this program every X seconds to "watch the devices continuously .... And that is not what I want.

Would be great if anyone can tell me if there is such a feature....

Regards

Joe

davidelang, 2017-12-01:
There is not such a feature, and there will not be. That's not the functionality 
that Tasmota is focusing on, and there isn't enough flash and ram in these 
devices to support that much additional complexity.

This is far from the first time this has been requestions. It may be that 
someone who wants this feature should make a fork of Tasmota and replace the 
MQTT functions with something that sends HTTP commands instead (and try to do it 
in a way that doesn't loose the ability to merge all the new sensors that 
tasmota keeps adding)

JoeWiseman, 2017-12-03:
Thanks for answering .... Not exactly what I WANTED to haer, but it does not hit me unexpected ....
Though I'm just taking profit from your great work and I can't even code a line opening a new branch would lead into a dead end ....
Though Im not familiar with the MQTT .... is it possible to get a "feedback" from the switch (over the "broker"), when using it along with MQTT?

closed 2018-05-07 by stale[bot]