arendst/Tasmota #1290: IrReceive Data is HEX, IrSend needs Integer

MatzeJoerling, 2017-12-01:
Hi, 
it needs some time to get it, Receive sends IR data to mqtt/console in HEX format.
I tried to send that same data out, but failed.
then i realized that i have to convert it form HEX to DEC (unsigned integer). 
Any idea why? Can irSend accept HEX as dataformat and send it out?
Makes IR handling much easier in my nodered.

regards M.

arendst, 2017-12-02:
But for now you managed to convert from hex to decimal in node-red.

Due to legacy compatibility a change to hex/hex kr decimal/decimal would be a problem.

MatzeJoerling, 2017-12-02:
hi,
must not be a change, could not irsend send it in dec and hex depends on parameter or # prefix.

regards M.

arendst, 2017-12-02:
Yes that would be the solution if users not already expected to send a number like 1234 as being hexadecimal.

Unlucky choices made in the past are hard to fix.

sammy2360, 2017-12-05:
so if this is legacy compatibility affecting IR and RF i take it its not the libraries used for IR and RF but deeper into the arduino core?

arendst, 2017-12-05:
No it's tasmota legacy.

sammy2360, 2017-12-05:
im just looking at the xdrv_snfbridge.ino file in particular this section 
```
  if (dataBuf[0] == '#') {
      dataBuf++;
      data_len--;
      radix = 16;
    }

    if (data_len) {
      code = strtol(dataBuf, &p, radix);
      if (code) {
        if (CMND_RFCODE == command_code) {
          sonoff_bridge_last_send_code = code;
          SonoffBridgeSendCode(code);
        } else {
          if (1 == payload) {
            code = pgm_read_byte(kDefaultRfCode + set_index) << 8 | pgm_read_byte(kDefaultRfCode + set_index +1);
          }
          uint8_t msb = code >> 8;
          uint8_t lsb = code & 0xFF;
          if ((code > 0) && (code < 0x7FFF) && (msb != 0x55) && (lsb != 0x55)) {  // Check for End of Text codes
            Settings.rf_code[0][set_index] = msb;
            Settings.rf_code[0][set_index +1] = lsb;
          }
        }
      }
    }
```
just trying to work out how to add a cmnd RF_CODE# for example, so that the sending of a hex value can be triggered from the topic of mqtt rather than the payload. so the topic would be `cmnd/sonoffBridge/RfCode#` and payload `8263CA`. only problem that may occur is using a # in mqtt may be dodgy.

arendst, 2018-04-02:
I'll change the way IRSend is processing the Data input field.

Starting with version 5.12.0i it will accept both decimal like 1234567 and hexadecimal like 0x12345A. So the change is that it now supports the standard C notation for hexadecimal being 0x. 

This means that the default mode is decimal with optional hexadecimal when prefixed with 0x or 0X.

closed 2018-04-02 by arendst