arendst/Tasmota #1292: DS18(B/X)20-(X) sorting / message issues when changing sorting behaviour

pascalsaul, 2017-12-01:
Could you please revisit the way you're ordering the DS18B20's in the messages/GUI once and for all :)? Every time you're changing the sorting of the sensors my domotica is confused because the sensors are shuffled (again). I think there is only one good way to address this issue. Use the unique address (ID) of every DS18B20 sensor instead of a DS18B20 and a follow number with the sorting. You can use this for both use cases: one sensor and multiple sensors.

Additional advantage: you know exactly which sensor is located where if you send out the message with the ID only. Also when you move a sensor from one device to another device the ID keeps the same. Also when you extend a device with now one sensor with another sensor it still keeps working. If you connect one sensor now and a second one you don't which one will be first in the new situation. 

So please use ID instead of DS18B20-<follow number> or DS18B20 for one sensor :)

mkh595, 2017-12-02:
Yes, this is not problem only for domotica but for all softwares. Fully agree with you, return DS18B20 as ID for each sensor has any sense. I am also using multiple DS sensors and is problem, when new sensor is added or when one is randomly not in report. Then order of sensors is completly different.

I would expect somethink like this, then we have not problem with order even sensors are not sorted:
```
"DS18x20":{"28441B450600000B":{"Temperature":17.1}, "28E4D9E10400001E":{"Temperature":17.2}....
```
More space can be saved when reported 'Temp' instead of 'Temperature'.

New structure in version 5.10 brings another negative side effect. Sensors are reported on same level like other information.
```
"Switch1":"ON","DS18B20-1":{"Id":"000004E1D9E4","Temperature":19.3},"DS18B20-2":{"Id":"000006451B44","Temperature":19.3}
```
Now cannot be used itteration through DS18x20 level. This code is flexible because we not need to know how many sensors are reported. But doesn't work in version 5.10

```
JsonObject& dss = root["DS18x20"];
for (JsonObject::iterator itdss = dss.begin(); itdss != dss.end(); ++itdss)
{
JsonObject& ds = dss[itdss->key];
    String _dsType = ds["Type"];
    String _dsAddr = ds["Address"];
    float _dsTemp  = ds["Temperature"];
```
For case where multiple sensors of one type (I guess it is only DS18x20) can be reported I would prefer separate level like was till version 5.9

arendst, 2017-12-02:
This is the first time sorting has been changed and kind of works. As sorting didn't work in previous versions the result were always different when adding new sensors.

Sorting is needed to have  a solid relation between the webpage and the json data.

Dynamic sensor addition is removed as it may sometimes fail to recognize a sensor leading to a missed entry in webpage and/or json.

For all of you fond of WordStar and WordPerfect I have made the previous driver still available as the Legacy version to be enabled in user_config.h.

pascalsaul, 2017-12-02:
Yes, but we don't like legacy... and that's is still not a long term solution. Why don't you sort on ID between the webpage and JSON data? Thereby everybody is using MQTT, don't we ;) Everytime the software detects an add or delete of a sensor it affects the numbering because it will be different (again).

You can use device/sensor/DS18B20/<id> for just one or multiple connected sensors. When one sensor failes it doesn't matter for the domotica or other software (like my InfluxDB with the sensor values) since there is no info for that particular ID.

Why not use <ID> instead of DS18B20-<number>? If you still want to sort, you can, but than on address ;)  For the webpage it doesn't matter either since it's displaying <ID> <Temp>.

This won't break anything and it's more universal and standardized for all software.

stefanbode, 2017-12-04:
I disagree to make the ID to an identifier and not like now to a property. If it is an identity JAVASCRIPT methods must know exactly the ID to extract the temperature and therefore you need one script for every sensor. Today with 1.2.3.4 you can use the same script for ALL of your devices to extract the temp from the JSON. Adding and removing sensors is NOT a common task. And if you do, you need to reassign it once, because during operation the assignment IS stable. If you like you can check on the backend, if the ID property matches and only assign if correct

pascalsaul, 2017-12-04:
To be honest I use a script (read: rule) for every sensor to make the 'script' stable. I think if you're using a generic script for reading your sensors it's a useless script because you don't use a specific sensor in your (use)case? Reading the temperature by ID is a huge advantage in my opinion so I can make the domotica (and software) more reliable!

jhlweb, 2018-02-18:
@arendst Hi Theo, why is the sorting of the DS18x20 array done in the first layer and not in a nested number array? Like this:
```
{
    "Time": "2018-02-18T00:15:21",
    "DS18B20": [
        {
            "Id": "0416A497BBFF",
            "Temperature": 85
        },
        {
            "Id": "0516A1CF96FF",
            "Temperature": 85
        }
    ],
    "AM2301": {
        "Temperature": 25.5,
        "Humidity": 31.3
    },
    "TempUnit": "C"
}
```
the -1 and -2 are now in JSONpath $.DS18B20.[0] and $.DS18B20.[1] this is much more like all other sensors. And still the sorting can be kept as needed for a stable environment. Also all DS18B20 sensors can be fetch in 1 call as opposed to the -1/-2/.. variant.

arendst, 2018-02-19:
@jhlweb will change it that way.

jhlweb, 2018-02-19:
@arendst oke thx. 
More of a thought. 
Every current sensor type has it's own branch/node. So isn't it normal to have a DS18B20 and a DS18S20 node.
But second why would you need the exact sensor type. I have a AM2302 and that is also under the AM2301 node and there is no way of knowing else then looking on the hardware component. It just reports Temperature and Humidity and a DS18B20 or DS18S20 reports an ID and a Temperature.
So calling them DS18x20 as before wasn't that strange a thing and have them under one node. 
Also if there was a setting to switch between ID and Address everyone who needs the type can use the Address to get the Type.