arendst/Tasmota #1294: Sonoff bridge not send fullset to domoticz

Tresett, 2017-12-02:
I'm connect RF Bridge to Domoticz, and take few transmitters.
Original Sonoff transmitter working fine, no problew.
Codes, from one other transmitter with 4 buttons - does not correct received in domoticz. Code is one.
This is info from RB bridge console.
Why packet with different High and Low values translate to one code?

```
13:58:25 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13290,"Low":410,"High":1290,"Data":"5597A8","RfKey":"None"}}
13:58:25 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5609384"}
13:58:36 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13300,"Low":420,"High":1280,"Data":"5597A8","RfKey":"None"}}
13:58:36 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5609384"}
13:58:44 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13300,"Low":420,"High":1270,"Data":"5597A8","RfKey":"None"}}
13:58:44 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5609384"}

```

arendst, 2017-12-02:
The code is not the result of High and Low.

The code is the payload send by the RF signal using timing pulses like a sync pulse length (Sync), a low pulse lenght (Low) and a high pulse length (High) of microseconds shown.

The differences you see in Sync, Low and High are normal for this kind of communication.

Tresett, 2017-12-02:
But I press 3 difference keys.

arendst, 2017-12-03:
Looking at your code value I see a possible problem value of 55xxxx. 55 is used to signal the end of comms between the RF receiver and the esp8266. 

I will prepare a change to not test on 55 until all values have been received. Hold on...

Tresett, 2017-12-03:
Tnx!
I can check at next weekend.

Tresett, 2017-12-10:
No changes.
Console messages after press 4 different key:
`10:02:16 MQT: tele/Sonoff_Bridge_1/STATE = {"Time":"2017-12-10T10:02:16","Uptime":20,"Vcc":3.194,"Wifi":{"AP":1,"SSId":"Lobakhin","RSSI":100,"APMac":"6C:3B:6B:21:83:91"}}
10:06:44 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13290,"Low":410,"High":1280,"Data":"55FE7E","RfKey":"None"}}
10:06:44 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5635710"}
10:06:49 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13290,"Low":410,"High":1280,"Data":"55FE7E","RfKey":"None"}}
10:06:49 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5635710"}
10:06:54 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13290,"Low":420,"High":1280,"Data":"55FE7E","RfKey":"None"}}
10:06:54 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5635710"}
10:06:58 MQT: tele/Sonoff_Bridge_1/RESULT = {"RfReceived":{"Sync":13310,"Low":420,"High":1270,"Data":"55FE7E","RfKey":"None"}}
10:06:58 MQT: domoticz/in = {"idx":15,"nvalue":0,"svalue":"5635710"}`

Domoticz log:
`2017-12-10 10:06:43.891 MQTT: Topic: domoticz/in, Message: {"idx":15,"nvalue":0,"svalue":"5635710"}
2017-12-10 10:06:49.724 MQTT: Topic: domoticz/in, Message: {"idx":15,"nvalue":0,"svalue":"5635710"}
2017-12-10 10:06:54.218 MQTT: Topic: domoticz/in, Message: {"idx":15,"nvalue":0,"svalue":"5635710"}
2017-12-10 10:06:58.395 MQTT: Topic: domoticz/in, Message: {"idx":15,"nvalue":0,"svalue":"5635710"}`

P.S. I can try to sniff radio with rdl_433. Intresting?

Tresett, 2017-12-10:
Sniff radio:
`trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
*** signal_start = 1152420, signal_end = 1243793
signal_len = 91373,  pulses = 134
Iteration 1. t: 212    min: 108 (80)    max: 316 (54)    delta 10
Iteration 2. t: 212    min: 108 (80)    max: 316 (54)    delta 0
Pulse coding: Short pulse length 108 - Long pulse length 316

Short distance: 103, long distance: 317, packet distance: 3314

p_limit: 212
bitbuffer:: Number of rows: 6 
[00] {25} 55 55 c0 00 : 01010101 01010101 11000000 0
[01] {25} 55 55 c0 00 : 01010101 01010101 11000000 0
[02] {25} 55 55 c0 00 : 01010101 01010101 11000000 0
[03] {25} 55 55 c0 00 : 01010101 01010101 11000000 0
[04] {25} 55 55 c0 00 : 01010101 01010101 11000000 0
[05] {9} 55 00 : 01010101 0
*** signal_start = 1929285, signal_end = 2012262
signal_len = 82977,  pulses = 121
Iteration 1. t: 210    min: 106 (71)    max: 315 (50)    delta 205
Iteration 2. t: 210    min: 106 (71)    max: 315 (50)    delta 0
Pulse coding: Short pulse length 106 - Long pulse length 315

Short distance: 105, long distance: 319, packet distance: 3317

p_limit: 210
bitbuffer:: Number of rows: 5 
[00] {25} 55 55 30 00 : 01010101 01010101 00110000 0
[01] {25} 55 55 30 00 : 01010101 01010101 00110000 0
[02] {25} 55 55 30 00 : 01010101 01010101 00110000 0
[03] {25} 55 55 30 00 : 01010101 01010101 00110000 0
[04] {21} 55 55 30 : 01010101 01010101 00110
*** signal_start = 2376812, signal_end = 2467163
signal_len = 90351,  pulses = 131
Iteration 1. t: 211    min: 107 (78)    max: 315 (53)    delta 5
Iteration 2. t: 211    min: 107 (78)    max: 315 (53)    delta 0
Pulse coding: Short pulse length 107 - Long pulse length 315

Short distance: 104, long distance: 318, packet distance: 3317

p_limit: 211
bitbuffer:: Number of rows: 6 
[00] {25} 55 55 0c 00 : 01010101 01010101 00001100 0
[01] {25} 55 55 0c 00 : 01010101 01010101 00001100 0
[02] {25} 55 55 0c 00 : 01010101 01010101 00001100 0
[03] {25} 55 55 0c 00 : 01010101 01010101 00001100 0
[04] {25} 55 55 0c 00 : 01010101 01010101 00001100 0
[05] {6} 54 : 010101
*** signal_start = 2875458, signal_end = 2946273
signal_len = 70815,  pulses = 100
Iteration 1. t: 207    min: 105 (60)    max: 310 (40)    delta 5
Iteration 2. t: 207    min: 105 (60)    max: 310 (40)    delta 0
Pulse coding: Short pulse length 105 - Long pulse length 310

Short distance: 105, long distance: 319, packet distance: 3318

p_limit: 207
bitbuffer:: Number of rows: 4 
[00] {25} 55 55 03 00 : 01010101 01010101 00000011 0
[01] {25} 55 55 03 00 : 01010101 01010101 00000011 0
[02] {25} 55 55 03 00 : 01010101 01010101 00000011 0
[03] {25} 55 55 03 00 : 01010101 01010101 00000011 0
*** signal_start = 4693159, signal_end = 4736703
signal_len = 43544,  pulses = 47
Iteration 1. t: 256    min: 134 (23)    max: 378 (24)    delta 5
Iteration 2. t: 256    min: 134 (23)    max: 378 (24)    delta 0
Pulse coding: Short pulse length 134 - Long pulse length 378

Short distance: 246, long distance: 0, packet distance: 247
`

Tresett, 2017-12-12:
Arendst, do you need more debug information?
Can I help smth else for this issue?

arendst, 2017-12-13:
As Tasmota can only act on decoded RF signals from the RF chip inside the bridge I'm afraid the bus stops here; if this chip does not recognize your signals Tasmota cannot act on it.

Tresett, 2017-12-13:
Can I debug data from serial (another chip)?
May be syslog?

arendst, 2017-12-13:
With logging set to 3 you would see this:
```
12:34:15 BRG: Received A4 30 2A 01 B8 04 F6 36 1F 13 55 
12:34:15 MQT: tele/bridge1/RFRECEIVED = {"RfReceived":{"Sync":12330,"Low":440,"High":1270,"Data":"361F13","RfKey":"None"}}
```
Where the BRG message is all data received from the RF bridge chip as defined in the ITead RF bridge protocol guide on their wiki https://www.itead.cc/wiki/File:RF_Universal_Transeceive_Module_Serial_Protocol_v1.0.pdf

Tresett, 2017-12-17:
10:10:17 BRG: Received A4 33 EA 01 9A 05 00 55 
10:10:21 BRG: Received A4 33 F4 01 9A 05 0A 55 
10:10:25 BRG: Received A4 33 EA 01 9A 05 00 55 
10:10:31 BRG: Received A4 33 F4 01 A4 04 F6 55 

Code 1, tlow and thight is different
Can I get fullset (A4 33 F4 01 A4 04 F6 55) in Domoticz?

arendst, 2017-12-17:
Just craft your own Fork and try to get the info to Domoticz. As it is timing you are after, prepare for different values for the same key.

Tresett, 2017-12-17:
May be can use another idx ... for example voltage or current for send fullset?