arendst/Tasmota #1295: magichome led power and brightness json?

donster22, 2017-12-02:
hi.

i have flashed sonoff-tasmota to a magichome led controller. thanks, it works great! i can't work ut how to send multiple commands to the controller to, for example, set colour and brightness at the same time.

i'm using node red and when i use the color wheel the console responds with:

    20:56:01 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":23,"Color":"0C0C3A"}

so, I am trying to send {"POWER":"ON","Dimmer":23,"Color":"0C0C3A"} as a json to my mqtt topic cmnd/sonoff/ with the command:

    cmnd/sonoff = {"POWER":"ON","Dimmer":23,"Color":"0C0C3A"}

but i get a {"Command":"Unknown"} response. does sonoff-tasmota support json and if so am i doing something wrong?

thanks, Don

arendst, 2017-12-02:
Tasmota does not use json for command entry. If multiple commands are needed start with command backlog followed by your commands seperated by a semicolon. See wiki commands page.

The color command will also set power to on and brightness as brightness is nothing more then a higher or lower value of red/green/blue. So in your colorwheel result of 0c0c3a the brightness is calculated as 23 procent. Changinf brightness will jusr re-calculate the correct color (with some obvious rounding margin).

tanloy, 2017-12-21:
I succeeded in flashing the firmware and can control colors using the console but I cant control brightness and color using homekit.   i can control on/off function but thats it.  Anybody willing to share their config.json for magichome?  thank you

mamieja, 2018-03-17:
Hi, same with me, can control everything via console on the webbrowser, but can't control it using the homekit app. It would be great, if somebody has succeeded in this and could share an excerpt of her / his config.json. Have a nice weekend!

Teie, 2019-05-29:
> hi.
> 
> i have flashed sonoff-tasmota to a magichome led controller. thanks, it works great! i can't work out how to send multiple commands to the controller, for example, to set colour and brightness at the same time.
> 
> i'm using node red and when i use the color wheel the console responds with:
> 
> ```
> 20:56:01 MQT: stat/sonoff/RESULT = {"POWER":"ON","Dimmer":23,"Color":"0C0C3A"}
> ```
> 
> so, I am trying to send {"POWER":"ON","Dimmer":23,"Color":"0C0C3A"} as a json to my mqtt topic cmnd/sonoff/ with the command:
> 
> ```
> cmnd/sonoff = {"POWER":"ON","Dimmer":23,"Color":"0C0C3A"}
> ```
> 
> but i get a {"Command":"Unknown"} response. does sonoff-tasmota support json and if so am i doing something wrong?
> 
> thanks, Don
> edit[typo]

 Hi, did you find the correct way to Turn on the Led?
 I'm a week trying to make a "night light" on the Node Red but I can't.