arendst/Tasmota #1298: Ntp not update

Tadzikj, 2017-12-03:
Hello.
sorry for my english but I am using a translator.

after connecting power to the sonoff with the Tasmot firmware the time update works correctly when the router is connected to the internetetm.
However, when the modem does not connect to the internet, sonoff loses time after being connected to the power supply and does not update it via ntp (because there is no internet). When the router is connected to the internet, sonoff does not update the time from ntp even after the Restart command.

00:02:00 MQT: tele / sonoff_E89A3C / UPTIME = {"Time": "1970-01-01T00: 02: 00", "Uptime": 1}
00:02:08 CMD: WebServer

Frogmore42, 2017-12-03:
Which version of Tasmota are you using?
I had one device (a Sonoff POW) that was running 5.8.0.  After some time it got confused and would never update the time (it was stuck in 1970, the beginning of time).  I even updated the FW to 5.9.0 via the web interface and saw that it was updated, but it still would never get the NTP time (but also did not display any error that I could see).  To solve the problem, I had to update the FW via the serial interface and that fixed it.  My guess is that there is some instability in the underlying SNTP code that only shows up in some cases.  All my other devices were working fine and they are all configured the same so I can't explain what was really wrong.

stefanbode, 2017-12-04:
NTP sync every hour at xx:01:01 no matter what time was there before. Additionally normally on reboot it should too if the NTP server response. The only possible way to "kill" it, is to point NTPSERVER1..4 to something that not works. @Frogmore42 the problem you described when the config get corrupted and there are arbitrary names for the NTP servers inside. I already asked for a CRC check, but declined. In these cases, a "reset 2" will do the job reliably.

Tadzikj, 2017-12-04:
Hello
I use 5.9.1j

I did a reboot, I even did reset 2 and then nothing :(

davidelang, 2017-12-04:
do a power cycle after the reset

Tadzikj, 2017-12-09:
witam 

dziekuje za pomoc. zmieniłem oprogramowanie na versje 5.10.0a i działa.
czasem zdarza sie nie zaktualizowac zegara. 

dziekuje

Tadzikj, 2018-02-08:
I have not paid attention lately but in fact my bridgrf after power outage does not update with ntp
I have another sonoff basic and it updates itself.

Frogmore42, 2018-02-11:
I found that if I unplug/power a device that is not updating the time for an extended period of time (30min is not too long for this, but it might need less).  When I plug it back in, it gets the correct time right away.