arendst/Tasmota #1299: 4ch pro, one channel self activates constantly

doppiaemme, 2017-12-03:
Hi, 

the first channel on a 4ch pro constantly turns on by itself.
the selected mode is inching with 0.5s timer.

I've tried flashing the board several times but it's still the same.
Can it be an hw issue? 

thanks

ottoplanet, 2017-12-21:
Hi,

I have somthing similar on my 4ch pro. Channel 4 is toggeling every 15 min in minute 27, 42,57 and 12. Even if I remove the power, self switching chennel 4 at minute 27, 42,57 and 12.
What can it be?

Thanks

ottoplanet, 2018-04-23:
My problem was solved. It seems, that I had activated the 433MHz remote command and my 4ch received an signal from my home automation system

Thanks

doppiaemme, 2018-04-23:
Ok this is interesting, 
how was the 433MHz channel activated? was it enabled by default after reflashing? and how did you disable it?
Still I wonder even of it was activated how it could pick up some signals that triggered it

closed 2018-06-22 by stale[bot]