arendst/Tasmota #1300: Adding pulse counter multiplier

roblad, 2017-12-03:
Hello,

I would like to ask you about the possibility to adding some extension or give indication of the place in the code, where I can change the counter of pulses - setting on CounterType - pulse with my requirements.

I would like report each 1  signal that to be the result of 400 real pulses received on GPIO. Saving each pulse received on GPIO will exhaust the capacity of the variable and write in to the flash.

Is it possible to add a variable in User_config i.e. pulse_devider in order to be able to count pulses in relation  in this variable.

Thanks

Frogmore42, 2017-12-04:
Yes, that is the variable.  But, to do what you want would require more RTC storage to keep track of the GPIO pulses that would then increment this pulse counter.  The pulse counter is an unsigned long so it can total up to over 4 billion, which is over 10 million L.  Generally, the answer would be to do the math in the system that is getting the data.

If you don't care about losing a partial (big) pulse (part of 10L) then you could keep track of the partial pulses in a static variable in this function (you would need to decide if you want to support more than one counter or not).  If you do care about preserving partial pulses then you would need to figure out a place to store the data and the process for doing it (probably in both Settings and RTCSettings).  I am not sure how much extra space there is, but you could always reduce to the number of counters by half and then use the other half for the divisor.  But, I don't know if this is something that would ever make it into the mainline.

roblad, 2017-12-04:
Hi,

Thanks for advice, but there is a litle bit diferenties. I Woudlike only react on pulse for every 410 or 4100 pulses, I can do that for my own repo only, but I do not know haw to do it - I do not know the code well and all dependencies between each ino, h and cpp files.

Regarding to that:
But, I don't know if this is something that would ever make it into the mainline.

It is good idea to have such counter multiplayer.

Rgs,

Frogmore42, 2017-12-04:
I am not an expert on the code as I have only been looking at it closely for a few days, but it looks well structured and has some comments in it.  Based on my reading of the responses to the issues, I don't believe this is a feature that many people would want and it has code and memory space implications, so I don't believe it would be something that will make it into the main line.

If all you want is one counter, you can take some short cuts in the implementation

Change this line:
`      RtcSettings.pulse_counter[index -1]++;`

To:
`      RtcSettings.pulse_counter[index -1]++;
       if (RtcSettings.pulse_counter[index -1] >= 4100) {
         RtcSettings.pulse_counter[index]++;
         RtcSettings.pulse_counter[index -1] = 0;
     }`

The code editor window is messing up the spacing, but it should give you an idea of what I mean.

You will then need to configure your device to have two counters.  The first one is for the pin that is attached to the meter.  The second one is for an otherwise unused pin that should be held high or low.  You will now get two counters in your status reports, one for the raw counts that should increment up to 4099 and the second one that will be a count of the times that the first one has done that.

This will work for single counter, but it is hard-coded to the actual divisor value.  Making it configurable would require more storage and more complexity.  This really is something that should probably be done in whatever is consuming the data.

roblad, 2017-12-04:
Hi,

see that :-)

https://github.com/stefanbode/Sonoff-Tasmota/issues/15

closed 2018-05-07 by stale[bot]