arendst/Tasmota #1302: New Sonoff Dual R2 Not activating relays from Web Interface

Snipercaine, 2017-12-04:
First, gotta say; Great Work!

I've been working with a Sonoff Dual.  I just upgraded it to 10.0.0a
Now the Button 0 and Button 1 pins activate the relays (awesome!), but the web interface toggle buttons do not (doh!). 
If I switch the configuration back to "05 Sonoff Dual" then the web interface buttons work, but of course Button 0 and Button 1 do not. 

I'm no programmer, but I'm going to look into the sonoff_template.h file and see if I can find the differences between the 05 Dual and the 39 Dual R2.  

-much thanks!

Frogmore42, 2017-12-04:
Do you actually have an R2 version of the Dual?
The HW is different between the original and the R2 version, so you have to use the one that matches the hardware that you have.

Snipercaine, 2017-12-04:
It says “ver 2.0”
https://www.dropbox.com/s/6raelivtbihpr1l/File%20Dec%2004%2C%202%2007%2025%20AM.jpeg?dl=0

It doesn’t work correctly with either the 05 Dual or the 39 Dual R2

arendst, 2017-12-04:
@Snipercaine That's still the "old" Dual where the BUTTON 0/1 header is not supported.

Snipercaine, 2017-12-04:
Ok
Thanks
Guess I’ll have to buy another and hope they send an R2

Snipercaine, 2017-12-04:
So on the R2 can you put the board in programming by connecting Button0 to ground?

qbitic, 2017-12-05:
Hi,
I want to know where to order a Dual so as to ensure that I'm buying the R2 version.
Anyone may help me?
Thank you very much

qbitic, 2017-12-05:
At this time I have the same board that Snipercaine has

Snipercaine, 2017-12-05:
@qbitic I ordered my first one from Amazon.  The box was crumpled so I'm not surprised it was the old version.  It looked like it had been kicked around a while.  
Today I ordered one directly from itead.  https://www.itead.cc/sonoff-dual.html
I figure if anyone is selling the newest version it might be them. 
I'll let you know when it arrives.  I've bought stuff from them before and the shipping is not fast.  I don't expect it to arrive for a few weeks.

qbitic, 2017-12-06:
> dual.html

Thaaaaanks!!!!!

tunino91, 2018-01-14:
@Snipercaine Heyy I just realized you are the Youtuber I recently have been following, small world. 
I am trying to find a good source for R2 as well for a painless firmware upload. Did your order turned out to be R2 after all? And how long did the shipping took if you can recall? Thanks

Snipercaine, 2018-01-14:
You’ve found my secret identity ;)

I did get an R2 when I ordered direct from iTead and specified in the comments for the order that I only wanted an R2. 
I don’t recall exactly how long it took, but it was faster than I expected. Maybe 2 weeks?...

tunino91, 2018-01-15:
@Snipercaine thanks a lot for the information and quick response!

pcb1962, 2018-04-08:
Thanks for this information folks. My Dual (bought last year but not used till now) says Version 2 on the board so I assumed it was an R2 - relay didn't work, found the solution here , thanks again.

kcmccombs, 2018-05-29:
Pcb1962 I am having this issue can you point me in the direction you found the answer thanks

pcb1962, 2018-05-29:
@kcmccombs Under 'Configure Module' on the web interface you need to select 'Sonoff Dual' not 'Sonoff Dual R2' unless you're sure you have a Dual R2 unit.

kcmccombs, 2018-05-30:
thanks for the fast reply i have already tried that.there is a R2 printed on it but it has early 2017 date on it.the leds change states but the relays do not i can manually make the relays and the work

Snipercaine, 2018-05-30:
@kcmccombs That sounds like what happened when I had the non-R2 but selected the R2 from the Module Type menu.  I know you said you tried that.  just try changing the module type and see if it does anything differently.  If that doesn't work maybe post the output from the console when you push the buttons.  Might be that the switchtopic is only sending mqtt and not controlling the relays directly. (that would happen if you have switchtopic2 I think).

closed 2018-05-30 by Snipercaine

kcmccombs, 2018-06-01:
thanks for the snipercanie when i try the other one i get the same thing no relays working.
but it seems that i have a new problem with the basic models when i try to re-flash it seems that nothing is being written to the sonoff but the software says it is i have been using the arduino  ide then i have tried the platformIO with the same results has any one had this problem thanks.

Frogmore42, 2018-06-01:
Look up cfg_holder.  You have to change it if you want the changed settings to be applied when you flash a device that was already configured.  But, the better way is to configure using the commands.

kcmccombs, 2018-06-01:
thanks for the reply is there a place where i can see how this is done.thanks