arendst/Tasmota #1304: Sonoff Basic and/or Dual with MQ2 Gas Sensor

Joeyhza, 2017-12-05:
I'm not sure if this has been asked and answered before, I did search the repo and couldn't find any results.

Is there support for the MQ2 gas sensor? I believe it is an analogue sensor.

ionciubotaru, 2017-12-06:
Yes, it's am analog sensor at 5V. You must use the analog pin for reading values. Be careful with 5V output as long as the esp8266 suports only 3V3

Joeyhza, 2017-12-06:
Thanks.

I think I'll just use the Pi GPIO to accomplish this. It supports 5v and no need for fine soldering on the Sonoff for analog pin :)

ionciubotaru, 2017-12-06:
What means "Pi GPIO" - raspberry pi GPIO ?
All raspberry pins are digital, not analog, and are not 5V tolerant

Joeyhza, 2017-12-06:
My apologies - I should have been more clear.

I have one of these

https://shop.pimoroni.com/products/automation-phat

Which gives the Pi Analog input. For the voltage issue the plan was to use a step up converter from 3v3 to 5v for the sensor. 

This was my plan B

Plan A was to connect the sensor directly to the Sonoff Basic, which would be ideal as its close to the gas source to be monitored. I'm completely unsure how to enable the analog input and initial research I've read talks about some serious soldering and my soldering skills aren't that great. Plus, I can't seem to find any pictures detailing how to expose the analog pin.

Joeyhza, 2017-12-08:
Just had a thought - considering the Sonoff Basics have 5v relays there must be a way to jumper in to a 5v feed on the PCB to piggyback on and power my MQ2 sensor :)

Will have to experiment

ionciubotaru, 2017-12-08:
don't even think about ...MQ2 is a hot sensor, it needs lot of power and  Sonoff 5V source is not designed to supply so much power - read the MQ2 datasheet

Joeyhza, 2017-12-08:
hmmm - there goes that idea :)

Thanks for the input

1043717432, 2017-12-29:
How to support MQ2 sensor?

ionciubotaru, 2017-12-30:
read the sensor's output analog values. Maybe you need a resistive divider because the output is 5V and exp8266 analog pin is not 5V compliant

1043717432, 2017-12-30:
@ionciubotaru  Thank you! Will Tasmota-web choose which project, you can read the simulation value!

ionciubotaru, 2017-12-30:
Instead of MQ2 (a hot and almost useless senzor) use BME680 from Bosch:
- it's small
- it's 4 in 1
- it's digital
- it's 100 times better than MQ2

1043717432, 2017-12-30:
@ionciubotaru  Price is too high！！ ：）

1043717432, 2017-12-30:
Mainly do not know which items to choose in the web?

ionciubotaru, 2017-12-30:
read this https://github.com/arendst/Sonoff-Tasmota/issues/1212

1043717432, 2017-12-30:
@ionciubotaru  I mean, in Tasmota how to set up to get MQ2 value, thank you!

ionciubotaru, 2017-12-30:
In `user_config.h` comment the following line:

#define USE_ADC_VCC                              // Display Vcc in Power status. Disable for use as Analog input on selected devices

choose WEMOS D1 mini as board type, than select analog GPIO

martinjuhasz, 2018-01-31:
Where does one find the analog input on a sonoff basic? is this exposed somewhere?

closed 2018-06-15 by stale[bot]