arendst/Tasmota #1305: MQTT broker failure fallback feature

pkaras3, 2017-12-05:
Hi,
I use Sonoff switches with SetOption11 = 1 to remotely toggle remote Sonoff switch from the Sonof Touch by one or double press by publishing a TOGGLE message in the buttontopic of the remote switch.
I have noticed that when the Sonoffs loose the connectivity to MQTT broker, they fall back to the primary functionality: one press just toggles the relay, which is very nice.
But in the situation when the remote switch is ON and MQTT broker fails it requires physical access to it to turn it off (but works as designed so far).
What about the fallback solution controlled by new SetOption when the connection with MQTT goes down:
 Case#:            Action: 
 1.                   do nothing (retain)
 2.                   power off when no connection to MQTT broker
 3.                   power on when no connection to MQTT broker
 4.                   toggle when no connection to MQTT broker
It would allow to power off the lights controlled by remote Sonoff switch when the switch is not easily accessible (hidden over fake ceiling or behind furniture) or even trigger some action.

Best regards,
Piotr

mkh595, 2017-12-05:
I noticed same issue if MQTT connection is not available. I have connected PIR sensor as Switch and it sends status with SwitchTopic via MQTT. If there is no connection to MQTT, PIR starts also control relay. That's problem because if I want switch light On manually via button, it is switch Off by PIR when PIR goes to state OFF. I would preffere if SwitchTopic is set, then switch cannot control relay also if MQTT is not connected. Otherwise there are two sources which wants to control same relay.

roguestreak, 2018-03-26:
Hi Theo, I second (OK, third) this request, I think it would be handy for a lot of situations;
One example-  I'm looking at switching my electric hot water with a POW, triggered by excess solar PV production, combined with a hardware timer set for daylight times AND a pre-morning slot.  Normally I don't want the morning slot to result in power unless I specifically tell it to with MQTT, so don't want an auto-on power setting, but if the MQTT server is unreachable for some reason then I DO want it to auto-on so we don't end up with a whole day of no water heating.

alfredopironti, 2018-06-03:
This looks related to #2100 

In my opinion, when SwitchTopic is set to a custom string, and the MQTT connection goes down, the behavior when the switch is triggered should be no action.

A custom SwitchTopic is used to send commands to a 3rd party, whose logic maybe completely unrelated with the relay attached to the sonoff. For example, I use a Sonoff to turn on/off a water pump. On this sonoff I also use a switch that triggers when a door is opened (which in turn turns on a light via HomeAssistant). Now, if the MQTT connection is lost, opening the door turns on the water pump, which is completely undesirable.

mawin01, 2018-06-06:
I have the same situation with some of my sensors, and I just want to let you know that I would very much welcome a solution or extension!

mawin01, 2018-06-20:
Me again. I didnt find a solution or workaround for that behavior. 

But, as I can not believe that we are alone with the setup : "a switch which is fully independant from the relay", is there a misconfiguration we did? 
Is the functionality "without a MQTT Connection the Relay was triggered by every switch which is connected to GPIO#" really a feature which we can not disable/avoid?

Can one of the developers or experts answer this question? Thanks in advance!

ascillato2, 2018-07-10:
Hi,

For the behaviour asked at the begining of this issue can be done now with the last firmware using rules

Please, read the wiki - rules. You can use (as example) the:

```
rule 1
rule on Mqtt#Disconnected do power1 off endon
```

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

DJack93, 2018-10-04:
I have similar issues for over a year. If the mqtt broker is down, all of my tasmota devices with motion sensors will switch the relays. 
My motion sensors are configured with a different SwitchTopic. All the logic is handled by NodeRed.  
I already tried:
- every switchtopic, switchmode, switch#, relay#, combination
- many Rules with Mqtt#Disconnected trigger

Unfourtunatly I had no luck. The rules doesn't seem to have an effect.
I'm on version 6.2.1

Is this something we can fix in the firmware?

ascillato, 2018-10-04:
Hi, is a misconfiguration.

Switchtopic will control your relays when there is no mqtt as by design

Switchtopic are for switches. 

For PIR sensors you need to use rules. 

Please, search in the issues for PIR in order to find rules examples. 

Thanks