arendst/Tasmota #1307: is it possible to disable the slampher RF's ability to turn light on/off?

stoneobscurity, 2017-12-05:
just a simple question, i couldn't find any info in the docs.

i have a slampher and it's flashed with 5.7.1, and i am getting interference from a neighbor that is turning on/off my light via RF, randomly. 

so i would like to be able to disable the RF entirely. (i use wi-fi to control it via home assistant.)

if this does not exist, then lets call it a feature request.

arendst, 2017-12-06:
It's a hardware feature where the rf chip is hardwired to GPIO0 normally used for a button.

To disable it you will have to find the track from the rf chip and GPIO0 and cut it.

stoneobscurity, 2017-12-09:
just flashed it with a fresh build of 5.10.0 with that set to zero. i'll let you know how it goes.