arendst/Tasmota #1314: SHT3x on Wemos D1 mini is working 5 minutes

fork04, 2017-12-06:
I see the readings above main menu :
SHT3X Temperature  21.8°C
SHT3X Humidity 51.9
after 5-7 minutes they are disapiring, nothing in logs, sometimes after restart they are back.

arendst, 2017-12-06:
Pls give the sht3x_v2 version a try and feedback your result.

fork04, 2017-12-08:
https://i.imgur.com/5m6EBlJ.png i`m on sht3x_v2 and i see nothing now

fork04, 2017-12-08:
https://github.com/wemos/WEMOS_SHT3x_Arduino_Library/tree/master/src - this library is working well and stable

arendst, 2017-12-08:
Pls give the V3 version based on Wemos a try.

fork04, 2017-12-08:
https://i.imgur.com/PcYikcz.png still not good

kskenyon, 2017-12-11:
Confirmed.  SHT3X_V3 not reading correctly.  

SHT3X: object
Temperature: 117917.6
Humidity: 65537

arendst, 2017-12-13:
OK. Wait for my SHT30 to arrive and I can test where it goes wrong...

kskenyon, 2017-12-15:
Similar problems with BMP280
It reads correctly for a couple of seconds when first connecting to the http interface then goes wacky and stays that way.

Could it be i2c related?

RaymondMouthaan, 2018-03-07:
Haii guys,
Not sure it this issue has been solved for you, but today I received my SHT30 shield for Wemos D1 mini (v2.1.0) and it runs just fine with Tasmota v5.12.0, it shows temp and hum with 1 decimal and measures for more than 30 minutes now ...

I guess this issue can be closed now ?