arendst/Tasmota #1316: Support for basic coding/logic rules like espeasy

thomasfa18, 2017-12-07:
Hi,
Is anthing like this currently supported?
https://www.letscontrolit.com/wiki/index.php/Tutorial_Rules

I tried searching but the wiki is very difficult to navigate

davidelang, 2017-12-07:
not in tasmota, you would put logic in your home automation tool

ChukXP, 2017-12-07:
Sometimes is better to do some important realtime operations directly in sonoff module to prevent any emergency cases becasuse of lost connection, slow feedback or automation tool hangs like "out of memory" in OpenHAB.

davidelang, 2017-12-07:
On Thu, 7 Dec 2017, ChukXP wrote:

> Sometimes is better to do some important realtime operations directly in 
> sonoff module to prevent any emergency cases becasuse of lost connection, slow 
> feedback or automation tool hangs like "out of memory" in OpenHAB.

True, but there is a very limited amount of flash and ram, so the choice is 
between supporting sensorts and adding local scripting capability.

Tasmota is not aiming at stand-alone devices, but rather at devices that are 
online and connected to a MQTT server.

If you are looking for something that runs stand-alone, you should pick a 
different project.

thomasfa18, 2017-12-07:
True, but the firmware is also modular at build time, giving users the ability to choose which components they want to be compiled or not ie. the user already have a choice about which sensors they want to support. The ram/flash usage is such a consideration that the firmware even supports a "minimal firmware". If anything this would indicate the foundation of adding a module for "basic logic rules" is already in place.

I understand that this discussion has drifted away from my original question of "is this currently supported", which is clearly a resounding "No!". However, if this is a feature multiple users are interested in then telling them to "go use something else" is stunting the development and expansion of this firmware.

mhavo, 2018-01-03:
> True, but the firmware is also modular at build time, giving users the ability to choose which components they want to be compiled or not ie. the user already have a choice about which sensors they want to support.

You don't have ALL the sensor connected ALL the time - so it would be wise to let the user to choose. **Modularity** is the key! *Local scripting* would make this a very robust FW for critical systems.

danielpondruff, 2018-03-09:
Totaly aggree. I would Love to see this Feature in tasmota

ascillato, 2018-06-02:
Hi

Now rules are in Tasmota. For further information please check the wiki. Thanks!

closed 2018-06-02 by ascillato