arendst/Tasmota #1317: no MQTT stat mess after cmnd

species5618, 2017-12-07:
repeatable on SonSoff POW and S20

MQTT topic config is
Topic = S20_1
Full Topic = /control/%topic%/%prefix%/

send MQTT     
topic /control/S20_1/cmnd/POWER
message: TOGGLE or ON / OFF

switch toggle and i only get backup topic
stat/SonoffS201/POWER 
back from the device not the expected /control/S20_1/stat/POWER

on the POW device, a config reset and manually reconfig resolved this issue, but no on s20

species5618, 2017-12-07:
I do however
Get the expected /control/S20_1/stat/POWER
On button push

species5618, 2017-12-07:
Both devices did not have tasmota previously 
We're programmed over serial port with 5.10

arendst, 2017-12-07:
Pls provide output of command ``status 0``

species5618, 2017-12-07:
This pow device current works as expected 
CMD: Status 0
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS = {"Status":{"Module":6,"FriendlyName":"SonSoff_pow1","Topic":"SonSoff_POW_1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":11,"Sleep":0,"BootCount":10,"SaveCount":100,"SaveAddress":"F6000"}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2017-12-01T13:55:52","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"MyHome","SSId2":"MyHome5","TelePeriod":60,"SetOption":"55800009"}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS5 = {"StatusNET":{"Hostname":"SonSoff_POW1","IPAddress":"172.17.2.25","Gateway":"172.17.2.240","Subnetmask":"255.255.255.0","DNSServer":"172.17.2.240","Mac":"2C:3A:E8:26:4F:3A","Webserver":2,"WifiConfig":2}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.2.250","MqttPort":1883,"MqttClientMask":"Sonoff_pow1","MqttClient":"Sonoff_pow1","MqttUser":"lee","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 07 10:27:49 2017","Local":"Thu Dec 07 11:27:49 2017","StartDST":"Sun Mar 26 02:00:00 2017","EndDST":"Sun Oct 29 03:00:00 2017","Timezone":1}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS8 = {"StatusPWR":{"Total":0.022,"Yesterday":0.010,"Today":0.012,"Power":0,"Factor":0.00,"Voltage":227,"Current":0.000}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS10 = {"StatusSNS":{"Time":"2017-12-07T11:27:49"}}
11:27:49 MQT: /control/SonSoff_POW_1/stat/STATUS11 = {"StatusSTS":{"Time":"2017-12-07T11:27:49","Uptime":11,"Vcc":3.230,"POWER":"ON","Wifi":{"AP":1,"SSId":"MyHome","RSSI":66,"APMac":"A0:F3:C1:99:2B:75"}}}

......

This s20 device does not
CMD: Status 0
11:32:34 MQT: /control/S20_1/stat/STATUS = {"Status":{"Module":8,"FriendlyName":"S20_1","Topic":"S20_1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:32:34 MQT: /control/S20_1/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":2,"Sleep":0,"BootCount":10,"SaveCount":68,"SaveAddress":"F6000"}}
11:32:34 MQT: /control/S20_1/stat/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2017-12-01T13:55:52","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:32:34 MQT: /control/S20_1/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"MyHome","SSId2":"MyHome5","TelePeriod":60,"SetOption":"55800009"}}
11:32:34 MQT: /control/S20_1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:32:34 MQT: /control/S20_1/stat/STATUS5 = {"StatusNET":{"Hostname":"S20_1","IPAddress":"172.17.2.40","Gateway":"172.17.2.254","Subnetmask":"255.255.255.0","DNSServer":"172.17.2.240","Mac":"5C:CF:7F:F9:17:B7","Webserver":2,"WifiConfig":2}}
11:32:34 MQT: /control/S20_1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"172.17.2.250","MqttPort":1883,"MqttClientMask":"S20_1","MqttClient":"S20_1","MqttUser":"lee","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
11:32:34 MQT: /control/S20_1/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 07 10:32:34 2017","Local":"Thu Dec 07 11:32:34 2017","StartDST":"Sun Mar 26 02:00:00 2017","EndDST":"Sun Oct 29 03:00:00 2017","Timezone":1}}
11:32:34 MQT: /control/S20_1/stat/STATUS10 = {"StatusSNS":{"Time":"2017-12-07T11:32:34"}}
11:32:34 MQT: /control/S20_1/stat/STATUS11 = {"StatusSTS":{"Time":"2017-12-07T11:32:34","Uptime":2,"Vcc":3.161,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MyHome","RSSI":100,"APMac":"70:4F:57:1E:CC:66"}}}

arendst, 2017-12-07:
It's caused by the fallback topic. The fallback topic is constructed from the MqttClient and overrules the FullTopic if Fallbacktopic = Topic.

If you change MqttClient from S20_1 to s20_1 you should be good to go.

species5618, 2017-12-07:
excellent thanks, it all looks good in the console window,

closed 2017-12-07 by species5618