arendst/Tasmota #1320: Best way to find Tasmota devices on my wifi?

TommySharpNZ, 2017-12-08:
My router doesn't reliably list all the attached wifi devices.
What is a better way to find my newly flashed tasmota devices on the network? I need to find out what their IP address is so I can configure them further after doing the first bit which is to enter the Wifi details into the admin page.

My PC is windows based and I've tried a few network scanning tool with not much success...

ionciubotaru, 2017-12-08:
install nmap for windows than:
`nmap -sP 192.168.x.*`

You will receive a list with all ip and macs in your LAN:

```
MAC Address: 60:01:94:84:61:0A (Espressif)
Nmap scan report for 192.168.1.3
Host is up (0.10s latency).
MAC Address: 60:01:94:00:54:9D (Espressif)
Nmap scan report for 192.168.1.4
Host is up (0.074s latency).
```

thrusty66, 2017-12-08:
For example with Fing - Network Tools for IOS and Android you can find all IP addresses in your own network. Quite simply and quickly.

SteveDinn, 2018-01-24:
One significant limitation of Fing is that it refuses to scan your network if there is no internet connection. I have all my automation devices on a separate wi-fi network that is not connected to anything external, save a machine with two NICs that bridges that network with my main, internet-connected wi-fi. I mean, I realize, it wouldn't be able to look up device types and MAC address information from its online database, but it's pretty annoying that it won't even scan.

ionciubotaru, 2018-01-24:
If you want to scan without internet:
`nmap -sP 192.168.x.*`

closed 2018-06-08 by stale[bot]