arendst/Tasmota #1321: H801 Device with Firmware 5.10.0

cradermacher, 2017-12-08:
Hi,
i have flasehd a H801 Device with the German version of the Firmware.
Everything works fine, an I can control the LEDs well.
I switched from Pam Mode to Color/Dimmer Mode with "Setoption15 1" and I can control the LEDs well.

Now, if I disconnect the Power from the module, and reconnect the Power again the LEDs starting in a white/blue color.
Getting the Status from color results in "0000000000" what is ok.  I can't control the module with commands color/dimmer. First I must reset the pwm mode with "setoption15 0" Now the LEDs switch off, an I can reset the color/dimmer mode.

If I disconnect the power from the module in the ppm mode everything works well.

Regards
Christian

arendst, 2017-12-13:
I cannot reproduce with my arilux but there might be an initialization omission in xdrv_light.ino

Pls could you try to change line 339 from:
```
  if (light_type < LT_PWM6) {           // PWM
    for (byte i = 0; i < light_type; i++) {
      Settings.pwm_value[i] = 0;        // Disable direct PWM control
    }
```
into:
```
  if (light_type < LT_PWM6) {           // PWM
    for (byte i = 0; i < light_type; i++) {
      Settings.pwm_value[i] = 0;        // Disable direct PWM control
      if (pin[GPIO_PWM1 +i] < 99) {
        pinMode(pin[GPIO_PWM1 +i], OUTPUT);
      }
    }
```

cradermacher, 2018-01-02:
Hi Theo,

thanks for the fix. I have testet it today, an it works for me. Now the H801 starts new, showing blue on the stripe and the going into the last state. Means off or showing the right color.

Thanks a lot for the very nice firmware. Its awesome.

Regards
Christian