arendst/Tasmota #1322: Delay when using a switch on GPIO14

joaohess, 2017-12-08:
Issue #1068 is closed but I'm having the same trouble.

I'm using Sonoff basic to control the lights of my house and I connected my wall switch to GPIO14/Ground. Whenever I toggle the switch it takes up to 10 seconds for my lights to turn on/off.

What can it be?

joaohess, 2017-12-08:
I'm still configuring my RaspberryPi with all my accessories, so there's no MQTT broker for now. They are working alone, only through the switch.

davidelang, 2017-12-08:
You are probably running into the bug in the arduino network library that 
prevent anything else from happening while it is attempting to connect to the 
mqtt server.

davidelang, 2017-12-08:
Then you need to disable the network, or disable mqtt so that the device isn't 
trying to connect to the MQTT server.

or just get the mqtt server online :-)

joaohess, 2017-12-10:
@davidelang that's it. I've disabled the MQTT on the sonoff and now it works instantly. Thank you!

closed 2017-12-10 by joaohess