arendst/Tasmota #1323: Old versions upgrade

fernac03, 2017-12-08:
i have one sonoff  with  version 3.2.13

is there  any place to download old versions  

Migrate to Sonoff-Tasmota 3.9.x
Migrate to Sonoff-Tasmota 4.x
Migrate to Sonoff-Tasmota 5.x

arendst, 2017-12-08:
See Releases at https://github.com/arendst/Sonoff-Tasmota/releases

fernac03, 2017-12-08:
one more question,

is safe to upgrade sonoff touch with this old pre compiled firmwares ?

davidelang, 2017-12-08:
define 'safe' :-)

it will probably work (unless you had defined CFG_HOLDER to a non-standard 
value), but there are lots of things that can go wrong.

your choice is to either try the multi-step upgrade, or just install the newest 
version via serial as if you never had Tasmota on the devices before.

fernac03, 2017-12-08:
i ask for safe because, is in production and i can't  have access to serial to install a newest version

davidelang, 2017-12-08:
it should work, but this is opensource software, it's not guaranteed (especially 
across so many versions).

your best bet is to setup a routine upgrade process so that you can upgrade in 
small steps, but you are still going to run into a bug someday

But you can't go back in time to avoid getting into the state you are in (I have 
some devices in a similar state that I haven't upgraded yet because I don't have 
easy access to them as well, I fully understand where you are at)

you have two choices

1. don't upgrade them (there is at least one security problem in software that 
old, but if they are only on a local network, and you are not a high value 
target, you may not care)

2. pull them all and upgrade via serial

3. try the staged upgrade. Best case, you upgrade everything and are current. 
Worst case, you have to pull them and upgrade via serial as per #2

there's no guaranteed way to upgrade them (just like there's no gurantee that 
upgrading Microsoft Windows won't break something, most people upgrade without a 
problem, but there are always reports of someone who's system is broken, and 
that's with millions of dollars spent in QA and upgrade testing)

jdpiguet, 2017-12-12:
You may have to test it in a lab before going into production...