arendst/Tasmota #1324: Broke the button, there's a way to switch to GPIO14

dboa, 2017-12-08:
I kinda broke the button and I wish to know how to overwrite from GPIO00 to GPIO14 on arduino IDE.

Is there a way?

davidelang, 2017-12-08:
you would need to change the module type to wemos, then you can define any GPIO 
as any function.

dboa, 2017-12-11:
@davidelang You're awesome.
It's doing exactly what I need.
Thank you so much

closed 2017-12-11 by dboa

morres83, 2018-04-23:
I have a question related to this topic:
Is the special button functionality (1-7 press for WIFI modes, reboot etc.) not statically mapped to GPIO0? Did it work with GPIO14?

Can the functionality be mapped to another switch (GPIO_SWT1)?
In my case the internal switch is not easy to access (physically), but I have an external switch (defined by GPIO_SWT1), which should trigger these events if necessary.