arendst/Tasmota #1327: Feature Request

Gary202, 2017-12-08:
Is it possible to add support to monitor current with addition of a Current Transformer SCT-013-000 and some circuitry with the ADS1115. I'm willing to pay for such support or fork. 
I'm looking to make a project to monitor my homes power usage. Similar to https://learn.openenergymonitor.org/electricity-monitoring/ct-sensors/how-to-build-an-arduino-energy-monitor-measuring-current-only

Genosse274, 2017-12-09:
The Sonoff Pow is capable of power monitoring by using the hlw8012. This feature seems to be already implemented.

Gary202, 2017-12-09:
The Sonoff Pow is only cabable of monitoring one device. Im talking about monitoring home mains up too 250 amps and 1 to 4 devices.

MatzeJoerling, 2017-12-10:
i use some of them https://de.aliexpress.com/item/5-45-A-45A-230V-50HZ-60HZ-Single-phase-Din-rail-KWH-Watt-hour-din-rail/32736928321.html (also available with > 100A) and tasmota switch / couner.
working well

arendst, 2017-12-13:
See #1151

Gary202, 2017-12-13:
That device is 214$us and doesn't support Posting to Mqtt. I need support for mqtt.

arendst, 2017-12-13:
It's not the device. It's my explanation regarding interrupt support in Tasmota.

roblad, 2017-12-19:
Hi

@MatzeJoerling 

Could you give more detailes on that ?

"i use some of them https://de.aliexpress.com/item/5-45-A-45A-230V-50HZ-60HZ-Single-phase-Din-rail-KWH-Watt-hour-din-rail/32736928321.html (also available with > 100A) and tasmota switch / couner.
working well"

MatzeJoerling, 2017-12-19:
Hi,

You could use any energy meter with S0 output.
https://www.ebay.com/sch/i.html?_from=R40&_trksid=p2380057.m570.l1313.TR12.TRC2.A0.H0.Xenergy+meter+s0.TRS0&_nkw=energy+meter+s0&_sacat=0

this is wired to my esp like this:
3.3V----R4.7k---(GPIOX,S0+)----(GND/SO+)

the meters sends you, for example, 2000 Impulse per KW/h. 
You can measure the length between 2 Impulses and get the actual power.
you could write the impulses to influx (deltas, not total) and do some grafana grafics on top.
may of them have a display where you can check your usage directly.

2 things, you could go with tasmota counter or switch.
Counter sends total impulse every 10 to 3600 seconds. the resolution not that good. but interrup driven - you will get every impulse. counter is stored in flash!
switches are queried every 50ms, S0 should be >30ms, you will miss some impulses. I have a patch available that queries the switch every 25ms. but you only get the impulses, if you backend misses some, they are lost.

you certainly will need some logic behind that, i have nodered,influxdb,grafana.
Looks like:

[image]
some nodered logic

[image]

all counters on top, hifi stuff below, watching tv till 8:20 ... 

[image]

all counters on nodered dashboard, not ony S0 power meters, also some of my knx actors can count power.

[image]

powering up the coffee maschine, then getting the coffee and a minute later the microwave for 30sec.
this is with switch, getting every impulse. with counter and 60sec teleperiod the same looks like:

[image]

but anyways, this setup is some years old, today i properly would go for this:
https://www.google.de/search?q=sdm630
and the small version sdm220 / sdm120 . they have modbus and rs485 which is working with espeasy since some days.
https://github.com/letscontrolit/ESPEasy/issues/205

regards M.

roblad, 2017-12-19:
Hi,

Thanks for well explanation but do you know something similar for 3 phase and with A and V mesaurement that can be operated by ESP and MQTT - even through tasmota :-)?

MatzeJoerling, 2017-12-19:
Hi, 
you want volt and ampere, right. sorry, i dont know. 
3 phase S0 counter is avail too.

roblad, 2017-12-19:
Yes,

I would like to get info for V and A as well.

orimate, 2018-05-15:
@MatzeJoerling Could you share your flow as text?

"https://user-images.githubusercontent.com/25864208/34146645-2556a656-e49b-11e7-96fd-007a6c211802.png"

MatzeJoerling, 2018-05-16:
hi, this is a flow with some function nodes special for my setup, perhaps not easy to clenup (what i have to do bevor sharing) or understand. 
Any special interest? perhaps that is easier to share. 

BTW, if you plan to do soemthing new, SDM630 support is on the feature list! #2735 
I have one also, go for that! Much better than S0.